ayushsharma82/ElegantOTA #2: Is that possible tho edit the html page ?

kbssa, 2019-05-20:
I was taking a look at the library files to see if it's possible to translate the texts and cahnge the logo, but the I have no ideia on how to changing the page that was made using some kind of encoding only  with numbers.

So I ask, is there a way to edit it ?

By the way, what a amazing library, only one line to do all the job. I was looking for something like that for days.

Thanks for your very good job.

Sorry about my poor english.

ayushsharma82, 2019-05-20:
Editing is not possible. But you can replace the webpage with your own. 

You can create your own webpage. Then you can use this [File2Raw](https://ayushsharma82.github.io/file2raw/) utility to convert it into Gzipped Hex data and replace it in `elegantWebpage.h` of `src` folder.

closed 2019-05-20 by ayushsharma82

kbssa, 2019-05-20:
> Editing is not possible. But you can replace the webpage with your own.
> 
> You can create your own webpage. Then you can use this [File2Raw](https://ayushsharma82.github.io/file2raw/) utility to convert it into Gzipped Hex data and replace it in `elegantWebpage.h` of `src` folder.

I got it.

Thank you very much for the fast reply.

bvishaloaf, 2021-05-06:
Can you share berif example that how to do update new page (replace new page elegantwebpage.h)
as i tried it crash not work