ayushsharma82/ElegantOTA #14: Add a retry button on failed upload.

DanielW0830, 2020-02-16:
A few times I had to hit back, then reload OTA page, set filename and click upload again to get the transfer to complete successfully.  (low wifi signal probably)

A retry button on a failed upload would be good.   :)

ayushsharma82, 2020-02-22:
That's was an intended feature to check if the module booted again after OTA or not.  I'll add it in next update.

HaegarDerWikinger, 2020-06-20:
Oh, I saw just now, the same idea then me. But I would like to go a step further and like to keep the selected file even if upload was ok.
Once I see that the program is not good enough, I change in Arduino IDE, compile and can upload again via ElegantOTA website. During development it happen very often that I need to change may just some values and upload again.
Thank You veru much and Best Regards
Michael

ayushsharma82, 2020-06-20:
I see. I'll add a 'back' button by tomorrow 👍. It will keep the file selected. But what if the IP of esp8266 has changed after OTA?

HaegarDerWikinger, 2020-06-21:
In my case it would not happen because I use fix IP address. 
If your approach allow changing IP, then have to check anyway for the new one.

ayushsharma82, 2020-06-21:
Added 'back' and 'retry' buttons for accessibility in ElegantOTA 2.2.0

Release: https://github.com/ayushsharma82/ElegantOTA/releases/tag/2.2.0

closed 2020-06-21 by ayushsharma82