ayushsharma82/ElegantOTA #19: "Batch" mode

brianmwebb, 2020-04-14:
Could you add some functionality to allow us to skip the elegant web page where we already know the bin file path and name, a "batch" update mode. We can then  can trigger the update when our application awakes from sleep and we know that it has an update waiting for it

ayushsharma82, 2020-04-14:
Do you mean a OTA process where we have same firmware for multiple devices?

brianmwebb, 2020-04-14:
Hi,

yes

A feature that will allow us to update multiple devices with the same bin,
perhaps we feed parameters of the bin file name and path and then the
function skips the webpage and moves directly to the OTA upload?

Regards

Brian Webb
0417 530972

ayushsharma82, 2020-04-14:
That is practically not possible with ElegantOTA, because first of all, we will need a separate update server to handle the query requests of multiple esp8266 devices ( for example, how will our esp8266 code know if this is a new update or it's the same firmware? ).

Sorry to say, this doesn't lie in the ballpark of ElegantOTA which is just a simple tool to upload computer to single device firmware updates.

If you would like this kind of functionality developed for your product/devices, then you can send me an email at asrocks5@gmail.com

Regards,
Ayush Sharma

closed 2020-04-16 by ayushsharma82

mindrunner, 2021-09-08:
Why would it not be possible to call `ElegantOTA` via curl?