ayushsharma82/ElegantOTA #20: Progmem

ddiabloss, 2020-04-18:
How to edit Webpage?

ayushsharma82, 2020-06-07:
Webpage source files are in the `ui` folder of the repo. Feel free to edit it. It's based on Vue.js.

HaegarDerWikinger, 2020-06-19:
Hi, 
I would like to mention, to hold the old file after uploading. During testing, I must do more often some uploads and had to refresh and search for the upload file.

Another question is, how to use this with ESP32? They use a different WiFi library and I couldn't get it to compile with together with the ElegantOta. Without no problem, just standard WiFi communication via ModBus TCP/IP with my Raspberry Master

For my ESP8266 it runs very nice. Thank You for developing and sharing.
Best Regards,
Michael

ayushsharma82, 2020-06-20:
@HaegarDerWikinger  There's an esp32 branch in this repository just for that, I faced some issues while sending the progmem page via its webserver so it didn't come out of development till now.

I'll test it again today, maybe the issue is fixed in esp32 arduino core.

ayushsharma82, 2020-06-21:
Added support for ESP32 with ElegantOTA 2.2.0

Release: https://github.com/ayushsharma82/ElegantOTA/releases/tag/2.2.0

closed 2020-06-21 by ayushsharma82