ayushsharma82/ElegantOTA #25: UI local development: work around /update/identity

qistoph, 2020-07-24:
Looking into some small modifications, I've found your local development option `npm run serve`. Which is really easy to setup and run.

However, upon opening the UI served locally I get an error that the JSON in `/update/identity` is invalid.

Opening that URL directly also shows the UI (HTML), instead of some JSON.

How would I work around that on the local development instance?

ayushsharma82, 2020-07-25:
`/update/identity` is a custom JSON response returned by the webserver on ESP8266 / ESP32.

To fix this in the local environment, We can just check the `process.env.NODE_ENV !== 'production'` and not make that HTTP request.

ayushsharma82, 2021-06-21:
Added with d6fd98a

closed 2021-06-21 by ayushsharma82