ayushsharma82/ElegantOTA #34: Upload status callbacks

zakutnya, 2021-01-10:
Thanks for the lib, it works magnificently right out of box, and what's more important — saves lots of time)

I've got a question about update process statuses callbacks.
E.g., i want to print some data on a screen after file upload is completed.

I went through the source files and found nothing suitable, am i missing something?

Thank you!

ayushsharma82, 2021-01-10:
@zakutnya You're welcome.
Sending a custom response back is not yet supported. 

A better alternative would be to use my WebSerial project :) 
https://github.com/ayushsharma82/WebSerial

zakutnya, 2021-01-10:
@ayushsharma82, Got it, thanks!
WebSerial looks very handy, will give it a try.

Looking forward to see callback features (in case they're on your list, of course))

Good luck!

closed 2021-01-10 by zakutnya