ayushsharma82/ElegantOTA #38: reload and get an Err_Connection_Timed_Out

J4e8a16n, 2021-03-13:
Hi,
I am experimenting with ESP32

https://github.com/ayushsharma82/ElegantOTA/blob/master/examples/ESP32_Demo/ESP32_Demo.ino
and I wonder why, after one minute or so, the web page is no more available.

If I press reload the page is reloaded but no more than a minute .
 
After that  if I reload I get an Err_Connection_Timed_Out .

If I reset the ESP32 everything comes back again …. 
Regards,
 
JPD

ayushsharma82, 2021-03-18:
@J4e8a16n  Hi, Thanks for trying out ElegantOTA.

Please check if any part of your code or library is closing the WebServer?

J4e8a16n, 2021-03-18:
It seems ....  I resetted the router to  defaults and everything is fine.

ayushsharma82, 2021-03-18:
Awesome! You may close the issue if it's fixed.

closed 2021-03-18 by J4e8a16n