ayushsharma82/ElegantOTA #48: LOGO changes

sonetonix-iot, 2021-05-25:
how to change custom logo instead of ElegantOTA logo...Please help me

ayushsharma82, 2021-06-21:
You can change this line: https://github.com/ayushsharma82/ElegantOTA/blob/master/ui/src/App.vue#L6

But try to attribute the library in your commercial projects to support open-source :)

closed 2021-06-21 by ayushsharma82

doanerock, 2021-06-21:
This would be great if it was added to the README.

sonetonix-iot, 2021-06-22:
Thank you for your support. I already change those lines, which you have
mentioned. but that will not work. there are no changes.

Let me know any other solution. Thanks in advance.

rbandres98, 2021-08-18:
> Thank you for your support. I already change those lines, which you have mentioned. but that will not work. there are no changes. Let me know any other solution. Thanks in advance.
> […](#)
> -- Thanks & Regards, *Naveen Ravichandran* *| Mobile : +91 9902678151 | E-mail : ***@***.*** ***@***.***> |* *Address: #1, SV Tower 3rd Floor, 2nd Cross, 1st Block, **Akshaya** Nagar, Ramamurthy Nagar, Bangalore - 560016*

Did you use `npm run build` command to recompile?

sonetonix-iot, 2021-08-18:
No...please let me know the details

rbandres98, 2021-08-24:
@sonetonix-iot you have to update elegantWebPage before flash your sketch again, please check [this comment](https://github.com/ayushsharma82/ElegantOTA/issues/56#issuecomment-904187339) that I made in another issue. Remember that you have to be know about [JavaScript Packages Basics](https://docs.npmjs.com/about-npm)

sonetonix-iot, 2021-09-20:
can guide me or any tutorial video..