ayushsharma82/ElegantOTA #52: Global Variable implemented in the header

GendoIkari, 2021-06-14:
Hi! I noticed that the global variable `ElegantOtaClass ElegantOTA` is declared and implemented in the header file. Implementing such variable in the header basically force the user to include it in the `main` `ino` file, and only there.

I would have expected the standard `extern` declaration + implementation in the `.cpp` for a global variable.

Is there a particular reason to do that? I would have considered this a simple oversight but I noticed that many arduino libraries are following this convention and I'm not a microcontroller developer. What is the benefit of forcing the user to include the library's header with a global object in the main file?

Thank you for your answer and your patience.

ayushsharma82, 2021-06-15:
If you see a global variable declared in a microcontroller library, Here are the following reasons:

1. The library is not supposed to have multiple instances.
2. It's user friendly to just include the header and start using the library.

All ESP core libraries are singletons, so that user doesn't accidentally create 2 or more objects to break the functionality. ( example WiFi class )

GendoIkari, 2021-06-15:
I'm not questioning the existence of a global variable, I'm questioning its definition in the header instead of the c/c++ way to declare global variables in the header with `extern` and defining it in the `cpp`.
If you define variable directly in the header you are only adding the constrain that I can't include that header more than once, nowhere, for any reason.
I can't have a function that takes a reference of an `ElegantOtaClass object` in another file, for example.

GendoIkari, 2021-06-15:
And be careful: you didn't create a singleton. Defining the variable in the header doesn't not automatically create a singleton. If I see correctly, you didn't protect the `ElegantOtaClass` `constructor`: the user can still create multiple instances if they want.

ayushsharma82, 2021-06-15:
@GendoIkari  It's a pretty generic implementation I follow with Arduino libraries. Feel free to improve it.

and including headers in multiple files should not be a problem as it has `#ifndef` preprocessor directive. So it will include that variable only for the first time.

GendoIkari, 2021-06-15:
Unfortunately, that's not correct. Every compilation unit is compiled by itself and then linked. In every compilation unit the `#ifndef` will be evaluated as true, and in every `.cpp` that includes the header the global variable will be redefined. The linker will complain that you are defining `ElegantOTA` multiple times.

However thank you for your time and your clarifications.

ayushsharma82, 2021-06-16:
I might not be that proficient in C++. However, let me know if we can split the library to header + cpp files and then declare the global variable in cpp?

ayushsharma82, 2021-06-21:
Not on the priority list.

closed 2021-06-21 by ayushsharma82

ayushsharma82, 2021-08-14:
Done with [v2.2.5](https://github.com/ayushsharma82/ElegantOTA/tree/2.2.5)