ayushsharma82/ElegantOTA #55: DOIT esp32 devkit v1 OTA v 2.2.5 not working

piotrret, 2021-08-17:
After upgrade from version 2.2.4 to 2.2.5 OTA not working.
Board is esp32 DOIT devkit v1. Server is responsing, but on the page is only ElegantOTA header, no checkbox firmware, no checkbox filesystem, no menu choose file.
[image]

rbandres98, 2021-08-17:
Same issue here

ayushsharma82, 2021-08-17:
Noted. I must have done a clerical mistake while refactoring code. 

You guys can use v2.2.4 till then it's the same.

Testing v2.2.5 right now.

ayushsharma82, 2021-08-18:
Fixed & Tested. Thanks for reporting.

closed 2021-08-18 by ayushsharma82

ayushsharma82, 2021-08-18:
For anyone who got stuck with v2.2.4 and can't update via serial to ElegantOTA v2.2.7, Issue was with the webpage but the underlying code is working. So Please follow these steps if your device is out of reach:

1. Make sure you have `cURL` installed on your PC
2. Open the directory where your new compiled firmware with ( ElegantOTA v2.2.7 or above ) is situated
3. Open terminal at that location
4. Execute this script `curl.exe -i -X POST -H "Content-Type: multipart/form-data" -F "firmware=@firmware.bin" http://192.168.4.1/update`
5. Wait for an "OK" message on the terminal.
6. Device is updated to ElegantOTA v2.2.7 and now you can resume OTAs via the webpage.