ayushsharma82/ElegantOTA #56: Update UI

nelsontechnologies, 2021-08-19:
After build for production in UI, what utility do you use to convert to hex to paste into elegantwebpage.h?  In the dist folder, i have favicon, index.html and js.  How do i merge these to the hex?  THank you in advance!

rbandres98, 2021-08-23:
Hello @nelsontechnologies .
The **elegantWebPage.h** file is automatically updated after use the command `npm run build` inside the **ui folder** of the project. The following is an image with the returned log in console after use this command.
[image]

ayushsharma82, 2021-12-20:
Solution provided by @rbandres98 . Thanks!

Closing ~

closed 2021-12-20 by ayushsharma82