ayushsharma82/ElegantOTA #60: ESP-01 Reset

AlexDuino, 2021-10-30:
With the versions 2.2.5 and 2.2.7 the ESP-01 and ESP-01s reset when access the web update ElegantOTA page.

Grey-Lancaster, 2021-10-31:
Bless your heart.  I am a noob and it was driving me nuts.  rolled back to 2.2.4 and it works fine
esp8266 nodemcu

`#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ElegantOTA.h>

const char* ssid = "****";
const char* password = "****";

ESP8266WebServer server(80);

void setup(void) {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  server.on("/", []() {
    server.send(200, "text/plain", "Hi! I am Papa Lanc.");
  });

  ElegantOTA.begin(&server);    // Start ElegantOTA
  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
}`

SergeSkor, 2021-11-29:
Confirm the same problem. Last good version is 2.2.4.
2.2.5 and above (up to 2.2.7) - esp8266 crashed on go to /update.

`--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x40218d91 epc2=0x00000000 epc3=0x00000000 excvaddr=0x800074dd depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffc60 end: 3fffffc0 offset: 0190
3ffffdf0:  00001df0 3ffffe60 80007461 40218da2  
3ffffe00:  000019f0 0000033e 0000033e 402135b6  
3ffffe10:  80007461 00000000 00000000 40100ee0  
3ffffe20:  3fff0e1c 00000007 40212a28 40213f28  
3ffffe30:  00000001 3ffffe64 00000004 40221c14  
3ffffe40:  80007461 00000000 3ffef01c 40213733  
3ffffe50:  3fff1604 3ffeec74 40212a28 40213f28  
3ffffe60:  402199c0 6470752f 00657461 872084a2  
3ffffe70:  3fff1200 00000000 3ffffeb8 40215de0  
3ffffe80:  80007461 3ffffeb8 3ffef08c 40205d34  
3ffffe90:  80007461 3ffef01c 3fff0484 40213cc3  
3ffffea0:  6470752f 00657461 87000000 40100e00  
3ffffeb0:  3fff0b5c 80ffff10 3ffffe00 40205d34  
3ffffec0:  80000001 40218d04 3ffffef0 40215bfd  
3ffffed0:  00000001 00000001 3fff05bc 401000e1  
3ffffee0:  3fff05bc 3ffeecb4 3fff05bc 40205d6c  
3ffffef0:  3fff0300 0010001f 801006e4 80002441  
3fffff00:  3fff05bc 3ffeecb4 3ffeec74 402084ee  
3fffff10:  3fff0b5c 0010001f 00ff1000 0000007f  
3fffff20:  80007461 4bc6a7f0 e0c49ba5 00000000  
3fffff30:  3ffeecb4 00000000 4bc6a7f0 00000001  
3fffff40:  00000001 3fff05bc 401006e4 00002441  
3fffff50:  00000000 3ffeec98 3ffeec74 3ffef254  
3fffff60:  00000001 3ffeec98 3ffeec74 40208607  
3fffff70:  00000000 00000000 3ffeef80 3ffef254  
3fffff80:  3fffdad0 00000000 3ffeee54 4020b3ad  
3fffff90:  3fffdad0 00000000 3ffef214 402086ba  
3fffffa0:  3fffdad0 00000000 3ffef214 40216fb8  
3fffffb0:  feefeffe feefeffe 3ffe8508 40101171  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

`

ayushsharma82, 2021-12-20:
Fixed with v2.2.8 - https://github.com/ayushsharma82/ElegantOTA/releases/tag/2.2.8

Thanks for reporting the issue.

closed 2021-12-20 by ayushsharma82