ayushsharma82/ElegantOTA #61: ElegantOTA writing to serial port

brownrb, 2021-11-08:
Hi

Thank you for writing this. It is nicely done and "elegant". 

However, I do have a suggestion. Can you please disable the writing to the Serial port. There is a message written to the Serial port when using ElegantOTA.

I cannot find where to disable this.

I do not believe that libraries should assume that the serial port is unused.

I have a few applications and projects that use the serial port for a variety of purposes, and having a message that I cannot disable coming from a library causes real issues for my applications/projects. I often start and stop the webserver service, and everytime it starts up I get this annoying message coming from ElegantOTA.

Thank you
Robert

ayushsharma82, 2021-12-17:
Hi Robert,
Typically you would want to use `Serial1` or SoftwareSerial for other applications where you need discrete control over which task can print data.

Currently, ElegantOTA doesn't have any compiler definition to turn off serial debug. I do think that critical tasks like device firmware upgrade should never have their debug output turned off.

ayushsharma82, 2021-12-17:
If you are GPIO restricted then you can fork the library and remove all the serial.print lines: https://github.com/ayushsharma82/ElegantOTA/blob/master/src/ElegantOTA.cpp#L82

brownrb, 2021-12-17:
respectfully, I disagree.
Please do not surmise what I need the srial port for. I had a simple request, which in my mind I framed politely and respectfully. 

And a blanket reply fo to a user request to at least provide a means to turn it off, that is really not a solution. If you are unwilling to listen to users then I would suggest that this applciation is not worthy of making into the "Arduino" list.

ayushsharma82, 2021-12-17:
I think you misunderstood my comment. It was intended to be helpful rather than sounding bullish.

My reasoning for suggesting `Serial1`/SoftwareSerial was because `Serial0` is also used by internal debug logging of ESP core. So one way or the other you will likely encounter some interference if something goes wrong.

Secondly, It's not a blanket reply, You said "I cannot find where to disable this", that's why I quoted the file where you can remove the lines because there is no actual way to remove it via a function call right now. ( and assumed a bit about the likely scenario in the back of my head )

>The bottom line is: There is no way right now to do it other than the 2 solutions that I "suggested". I've got some pending bugs to fix in this library before I can hop on to new additions.

closed 2023-09-23 by ayushsharma82