ayushsharma82/ElegantOTA #62: [HTTP Error] Not Found

mrlightsman, 2021-11-12:
I used the esp8266 demo as offered, only entering my wifi credentials.

Webserver loads fine.  /update takes me to the update page.   I navigate to the bin file for upload and click open.

The task bar progress to 100%, but then I get an [HTTP ERROR] Not Found message.  The upload has not worked.

I am using a nodemcu 12F board with the latest arduino ide.

What have I done wrong?  Thanks.

Johboh, 2021-11-16:
I'm experiencing the same issue. Downgrading to 2.2.4 solves the issue (the version before the refactoring)

mrlightsman, 2021-11-17:
Thank you. That did the trick. The new problem I have is when I try to set ElegantOTA.setID(“xyz”) the update webpage just spins. The elegant ota image is there, but the file upload section never loads.  Any thoughts on that?  I have the potential of a lot of devices on one network. I really want to utilize the id marker to help keep them straight. Thanks.

Johboh, 2021-11-17:
I'm experiencing the same issue with the setID. It works sometimes on some devices, but mostly it doesn't. I'm not aware of a workaround for that one.

mrlightsman, 2021-11-17:
Ok. Thanks. Let’s hope for a future fix.

closed 2021-11-17 by mrlightsman