ayushsharma82/ElegantOTA #63: crash on sart

KavehSons, 2021-11-26:
hi, 

I've tried to follow the example on[ this link](https://randomnerdtutorials.com/esp32-ota-over-the-air-arduino/) on my wroom 32 dev board , however after uploading its continue to crash with the following result 
ant help really appreciated 

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
[0$⸮LW+QW⸮Y.\&⸮KZX⸮K,\]⸮,'⸮LLW⸮⸮Y](\]⸮⸮equencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    49][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 0 - WIFI_READY

Brownout detector was triggered

```

ayushsharma82, 2021-12-17:
Hi @KavehSons,

`Brownout detector was triggered` is a sign of inadequate power supply. Please make sure you power your ESP with at least 3.3v - 1A.

closed 2021-12-20 by ayushsharma82

turboishack, 2024-10-17:
You can try different cables and power sources, but all you need is a 100uF 16V capacitor between 3.3V and GND on your ESP32 dev board.