ayushsharma82/ElegantOTA #64: setID is not working.

SergeSkor, 2021-11-29:
have long time using project with ElegantOTA, it works good on ver 2.2.4
Today have decided to add setID() functionality, and find out **/update** page became stuck, always spinning the circle under the page header. Micro is still working, no crash. Have removed setID() - **/update** works again.
btw, have updated to newer versions, esp8266 crashed disregarding with no setID() or without, seems like it is another bug.

Have left my project with old 2.2.4 version without setID().
:-(

SergeSkor, 2021-11-30:
I have played a little bit with ElegantOTA.h (version 2.2.4) editing directly getID() function.
`       
       String getID(){
            String id = "";
            #if defined(ESP8266)
                id = String(ESP.getChipId()); // <<--- editing this line!
            #elif defined(ESP32)
                id = String((uint32_t)ESP.getEfuseMac(), HEX);
            #endif
            id.toUpperCase();
            return id;
        }
`
It fails when not decimal digits are included in mentioned line.
When I have substitute **id=String("123456");** - it works ok,
But with **id=String("12345A");** - it endlessly rotates the circle when going to **/update**.

ayushsharma82, 2021-11-30:
@SergeSkor I'll get back to you with a fix. Are you willing to test out some changes?

SergeSkor, 2021-11-30:
> @SergeSkor I'll get back to you with a fix. Are you willing to test out some changes?

Sure!
Seems like extra quotes required, **server->send(200, "application/json", "{\"id\": \""+String("\"65432A\"")+"\", \"hardware\": \"ESP8266\"}");** is working well.

ayushsharma82, 2021-11-30:
Please checkout this branch: https://github.com/ayushsharma82/ElegantOTA/tree/id-fix
This is a fork for v2.2.7, please look out for 3 things:
1. See if the ESP8266 crash is fixed
2. Check if the webpage loads fine with the default ID
3. Check if the webpage loads fine with custom ID ( `setID()` ).

I think it's more related to how ID is generated at the initialization.

SergeSkor, 2021-11-30:
> Please checkout this branch: https://github.com/ayushsharma82/ElegantOTA/tree/id-fix This is a fork for v2.2.7, please look out for 3 things:
>
> 1. See if the ESP8266 crash is fixed
> 2. Check if the [...]

Still crashed on ESP8266, whenever with default ID or with setID("123456").

ayushsharma82, 2021-11-30:
Eh Okay... What does the ESP exception decoder say?

SergeSkor, 2021-11-30:
Continue to play with 2.2.4 version, which is not crashing on esp8266.
Seems like these minor fixes are working to allow setID () with non-digit characters.
`        void setID(const char* id){
            _id = String("\"") + id + "\"";
        }

       String getID(){
            String id = "";
            #if defined(ESP8266)
                id = "\"" + String(ESP.getChipId()) + "\""; 
            #elif defined(ESP32)
                id = String((uint32_t)ESP.getEfuseMac(), HEX);
            #endif
            id.toUpperCase();
            return id;
        }
`

ayushsharma82, 2021-11-30:
Hmm interesting. With your fixes in place, can you go to this URL: `http://<your esp IP>/update/identity` and paste the response here?

SergeSkor, 2021-11-30:
> Eh Okay... What does the ESP exception decoder say?

Here is the error:

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x40218de9 epc2=0x00000000 epc3=0x00000000 excvaddr=0x800074dd depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffc50 end: 3fffffc0 offset: 0190
3ffffde0:  00001a60 3ffffe50 80007461 40218dfa  
3ffffdf0:  00001968 0000032d 3ffeec84 402135e2  
3ffffe00:  80007461 00000000 00000000 40100ee0  
3ffffe10:  3fff03dc 00000007 40212a30 40213f80  
3ffffe20:  00000001 3ffffe54 00000004 40221c6c  
3ffffe30:  3ffffe9c 00000000 3ffef02c 4021375f  
3ffffe40:  3fff12bc 3ffeec84 40212a30 40213f80  
3ffffe50:  40219a18 6470752f 00657461 87215ff8  
3ffffe60:  3ffe8eb9 00000007 3ffffe90 40215e38  
3ffffe70:  3ffffe9c 3ffffe90 3ffe8e04 40205d34  
3ffffe80:  3ffffe9c 3ffef02c 80007461 40213df9  
3ffffe90:  6470752f 00657461 87feec84 3fff1200  
3ffffea0:  0028002f 80000000 64692200 22203a22  
3ffffeb0:  80ff1100 36353900 00303930 80205d34  
3ffffec0:  3fff0300 0019001f 80fffef0 40215c55  
3ffffed0:  00000001 00000001 3fff0604 401000e1  
3ffffee0:  3fff0604 3ffeecc4 3fff0604 40205d6c  
3ffffef0:  3fff1100 0010001f 801006e4 80004443  
3fffff00:  3fff0604 3ffeecc4 3ffeec84 402084f6  
3fffff10:  3fff119c 0010001f 00ff0e00 0000007f  
3fffff20:  80007461 4bc6a7f0 73333333 00000000  
3fffff30:  3ffeecc4 00000000 4bc6a7f0 00000001  
3fffff40:  00000001 3fff0604 401006e4 00004443  
3fffff50:  00000000 3ffeeca8 3ffeec84 3ffef298  
3fffff60:  00000001 3ffeeca8 3ffeec84 4020860f  
3fffff70:  00000000 00000000 3ffeef90 3ffef298  
3fffff80:  3fffdad0 00000000 3ffeee64 4020b3b5  
3fffff90:  3fffdad0 00000000 3ffef258 402086c2  
3fffffa0:  3fffdad0 00000000 3ffef258 40217010  
3fffffb0:  feefeffe feefeffe 3ffe8508 40101171  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld

SergeSkor, 2021-11-30:
> Hmm interesting. With your fixes in place, can you go to this URL: `http://<your esp IP>/update/identity` and paste the response here?

Without my fix: **{"id": 4956090, "hardware": "ESP8266"}**
With my fix: **{"id": "4956090", "hardware": "ESP8266"}**

SergeSkor, 2021-11-30:
> > Hmm interesting. With your fixes in place, can you go to this URL: `http://<your esp IP>/update/identity` and paste the response here?
> 
> Without my fix: **{"id": 4956090, "hardware": "ESP8266"}** With my fix: **{"id": "4956090", "hardware": "ESP8266"}**

Actually, due to fact old version (2.2.4) is working well with default ID, fixing getID() function is kind of extra, just for consistency, to always have quotes. But setID() is needed in order to accept non digital ID.

ayushsharma82, 2021-11-30:
I'm so puzzled right now haha. The escaped quotes are already there before the `/identity/response` response string concatenates. After concatenation, I don't know where the hell they are gone.

I'll check this with my ESP later on. Also, We will have to work on getting the latest commit fixed because the whole codebase was refactored with v2.2.5

SergeSkor, 2021-11-30:
> I'm so puzzled right now haha. The escaped quotes are already there before the `/identity/response` response string concatenates. After concatenation, I don't know where the hell they are gone. [...]

Just add another pair of quotes, haha :-).
Ok, sure, will help as much as I can.  For now will stay with 2.2.4 with my fix...

ayushsharma82, 2021-12-18:
Hi @SergeSkor,

Pushed some commits to the id-fix branch. I've got the example running, but can you test it with your code as well?

Thanks for the help!

SergeSkor, 2021-12-18:
Seems like it works, thank you! Did not really test filesystem update, have updated firmware only, it was no problem. SetID() is also working correct. Sent from Mail for Windows From: Ayush SharmaSent: Saturday, December 18, 2021 11:17 AMTo: ayushsharma82/ElegantOTACc: SergeSkor; MentionSubject: Re: [ayushsharma82/ElegantOTA] setID is not working. (Issue #64) Hi @SergeSkor,Pushed some commits to the id-fix branch. I've got the example running, but can you test it with your code as well?Thanks for the help!

ayushsharma82, 2021-12-18:
@SergeSkor Thanks for testing! I'll merge it soon.

ayushsharma82, 2021-12-20:
Merged. Release: https://github.com/ayushsharma82/ElegantOTA/releases/tag/2.2.8

closed 2021-12-20 by ayushsharma82