ayushsharma82/ElegantOTA #65: Don't flash when IDs are set.

ninjaju, 2021-12-01:
Hello,

Thank you for your great job!

I found an issue.

I'm with 2.2.6.

I set ID and password on the /update page with this line : AsyncElegantOTA.begin(&server, admin, admin);    // Start ElegantOTA

When I flash with the new firmware the update is not set. If I remove the ID and flash, it's working fine.

Thank tou for your support. :)

ayushsharma82, 2021-12-18:
Hi @ninjaju ,
I've added some fixes in [id-fix](https://github.com/ayushsharma82/ElegantOTA/tree/id-fix) branch, can you please test those out and see if it fixes your issue? You can download that branch and replace it with the existing ElegantOTA folder in your Arduino libraries folder.

ayushsharma82, 2021-12-20:
Fixed with v2.2.8 - https://github.com/ayushsharma82/ElegantOTA/releases/tag/2.2.8

Thanks for reporting the issue.

closed 2021-12-20 by ayushsharma82