ayushsharma82/ElegantOTA #66: OTA don't work with hardware timer

jlfrucot, 2021-12-09:
Maybe not an issue !
The use of hardware timer blocks OTA update. This may solved by settings this timer in IRAM with IRAM_ATTR like in this example :

`int16_t tempsRestant = timeToGo;
hw_timer_t *Chrono_timer = NULL;
portMUX_TYPE timerChronoMux = portMUX_INITIALIZER_UNLOCKED;
bool ChronoIsRunning = false;
IRAM_ATTR void Chrono_decremente()
{
    portENTER_CRITICAL_ISR(&timerChronoMux);
    tempsRestant--;
    portEXIT_CRITICAL_ISR(&timerChronoMux);
}

IRAM_ATTR hw_timer_t * Chrono_init(uint16_t time)
{
    tempsRestant = time;
    Chrono_timer = timerBegin(2, 80, true);
    timerAttachInterrupt(Chrono_timer, &Chrono_decremente, true);
    timerAlarmWrite(Chrono_timer, 1000000, true);
    timerAlarmEnable(Chrono_timer);
    return Chrono_timer;
}
`

ayushsharma82, 2021-12-17:
Yeah, DFU ( device firmware upgrade ) is a CPU intensive task. I'll soon add callbacks to ElegantOTA so that users can pause all timers and tasks gracefully before the OTA starts.

hitenlulla, 2022-02-13:
Hey, did you find a solution to this problem? My device highly depends upon Hardware Timer and I want to add the feature of OTA updates. @jlfrucot @ayushsharma82

ayushsharma82, 2023-10-17:
@hitenlulla  Added callbacks in V3. These can be used to stop timers before update starting downloading.