ayushsharma82/ElegantOTA #67: version 2.2.6 does not show OTA success after upload of new SW version

dolhastelian, 2021-12-12:
The SW is loaded correctly and shows 100% but it does not show the OTA success page

dolhastelian, 2021-12-12:
[image]

ayushsharma82, 2021-12-17:
@dolhastelian  Can you tell us more about your ESP model and board config?

dolhastelian, 2021-12-17:
Yes, I tested on esp-01 and Node MCU v2 modules, flashing is successful but not reported in the Web page.

ayushsharma82, 2021-12-18:
Hi @dolhastelian ,
I've added some fixes in [id-fix](https://github.com/ayushsharma82/ElegantOTA/tree/id-fix) branch, can you please test those out and see if it fixes your issue? You can download that branch and replace it with the existing ElegantOTA folder in your Arduino libraries folder.

ayushsharma82, 2021-12-20:
This should be fixed with v2.2.8 - https://github.com/ayushsharma82/ElegantOTA/releases/tag/2.2.8

Please let me know if the issue still persists after updating the library.

dolhastelian, 2021-12-23:
Hello Ayush!,
I tested the latest version and it works!
Can you make an update also for the AsyncElegantOTA ? Thank you.

[image]

ArnoldGoat, 2022-11-28:
Could we have 2.2.8 installed in VSCode PlatformIO. Latest is 2.2.7 which has the problem described on a Wemos Mini ESP8266. I don't know how to check 2.2.8 as the names don't seem to match (AsyncElegantOTA - ElegantOTA)

ayushsharma82, 2023-10-17:
Permanently fixed with V3.

closed 2023-10-17 by ayushsharma82