ayushsharma82/ElegantOTA #69: Disable Filesystem Upload

dandud100, 2021-12-21:
Hello, is it possible to disable or hide the filesystem update?

I use LittleFS for my projects so this is not useful for me.
Thanks.

ayushsharma82, 2021-12-21:
Nope, the webpage is pre compiled. By the way it can be used to update
LittleFS too.

dandud100, 2021-12-29:
How do i define it for LittleFS?

rbandres98, 2022-01-19:
Hello, @dandud100 . If you want to hide filesystem radio button you have to modify UI project by your way, you can find this at the following line: https://github.com/ayushsharma82/ElegantOTA/blob/master/ui/src/App.vue#L91

dandud100, 2022-01-19:
> Hello, @dandud100 . If you want to hide filesystem radio button you have to modify UI project by your way, you can find this at the following line: https://github.com/ayushsharma82/ElegantOTA/blob/master/ui/src/App.vue#L91

If I could define elegantota for littlefs it would be useful for me but I don't know how to do this. It is not mentioned anywhere.

ayushsharma82, 2023-09-23:
This feature is in [ElegantOTA Pro](https://elegantota.pro) now.

closed 2023-09-23 by ayushsharma82