ayushsharma82/ElegantOTA #71: Prevent deleting Files in LittleFS Filesystem when uploading OTA

dandud100, 2022-01-16:
How to prevent that the files on the LittleFS gets deleted when uploading an ota?

ayushsharma82, 2022-01-19:
Hi @dandud100 ,

You can try compiling the binary with the `Erase Flash > Only Sketch` option. Let me know if that keeps the LittleFS contents. 

[image]

dandud100, 2022-01-19:
> Hi @dandud100 ,
> 
> You can try compiling the binary with the `Erase Flash > Only Sketch` option. Let me know if that keeps the LittleFS contents.
> 
> [image]

I did this and it erased the littlefs content

ayushsharma82, 2023-10-17:
Tried on my end. It didn't delete LittleFS partition.

closed 2023-10-17 by ayushsharma82