ayushsharma82/ElegantOTA #72: OTA update via platformio not working and Preferences.h questions

PPKGroup, 2022-01-20:
Hi,

I'm using platformio (on VSC) and Win10 for programming of my device.
In my platformio.ini file I have placed the following code:
[env:WiFi]
upload_protocol = espota
upload_port = 10.100.8.60

When I try to build and upload, I get the following error:

_Uploading .pio\build\WiFi\firmware.bin
14:18:41 [DEBUG]: Options: {'esp_ip': '10.100.8.60', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 48565, 'auth': '', 'image': '.pio\\build\\WiFi\\firmware.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
14:18:41 [INFO]: Starting on 0.0.0.0:48565
14:18:41 [INFO]: Upload size: 1284432
Sending invitation to 10.100.8.60 ..........
14:18:41 [ERROR]: No response from the ESP
*** [upload] Error 1_

If I browse to the device e.g. http://10.100.8.60/update, the ElegantOTA web page is displayed.

Any ideas on how to get this to work through the platformio environment?

I read somewhere on the RNT website that ElegantOTA overwrites the spiffs and Preferences (NVS) flash storage.
In platformio is there a way of limiting where ElegantOTA writes to?

Thanks,
Neil.

viktak, 2023-09-13:
I would also like to have this feature implemented... It was sooo easy to use it before Asynwebserver...

normen, 2023-09-22:
For me it works like this with the extra script, note the URL:
```
extra_scripts = platformio_upload.py
upload_protocol = custom
upload_url = http://192.168.123.123/ota/upload
```

ayushsharma82, 2023-10-19:
@mindsuru The internal API of ElegantOTA indeed got changed to provide a much better & stable experience. Your fix looks promising! It was a community PR that added it in past that's why it wasn't updated with V3.

If you don't mind, can you open a PR that fixes the `platformio_upload.py` script?

mindsuru, 2023-10-19:
Thanks,
and okay, I'll give it a try. I'm not that familiar with GitHub yet, but it should work out somehow. I'll read up on it.

viktak, 2023-10-30:
@mindsuru For me the version you have [here](https://github.com/ayushsharma82/ElegantOTA/issues/72#issuecomment-1771361711) does not work. It gives the following error, then quits:
```
....
Uploading .pio\build\esp32\firmware.bin
Start-Request fehlgeschlagen 401
```

Seems to me that it cannot find the OTA page... This is the relevant part of my `platformio.ini`:
```
extra_scripts = 
    pre:../../scripts/preIncrementBuildNumber.py
    platformio_upload.py
upload_protocol = custom
upload_url = http://192.168.123.137
```
As you can see I have an extra script running, but I don't suppose that would be an issue as the behavior is that same when I remove it.
If I can help testing any new version of your script, don't hold back :)

bartoszzajac, 2023-11-21:
@mindsuru @ayushsharma82 When can we expect PR and merge?

mindsuru, 2023-11-22:
@viktak is your device reachable with this ip you‘ve choosen in your platformio.ini file?

ayushsharma82, 2023-11-22:
> @mindsuru @ayushsharma82 When can we expect PR and merge?

It has been merged but yet to be released.

ayushsharma82, 2023-11-22:
> @viktak is your device reachable with this ip you‘ve choosen in your platformio.ini file? 
> 
> 

I guess he's using authentication that's why response code was '401' which means 'Unauthorized'. The script might need adjustments to work with authentication enabled.

viktak, 2023-11-22:
Yes, the device is reachable, all other functions of the asyncwebserver and elegantota are all working.

viktak, 2023-11-22:
> > @viktak is your device reachable with this ip you‘ve choosen in your platformio.ini file?
> 
> I guess he's using authentication that's why response code was '401' which means 'Unauthorized'. The script might need adjustments to work with authentication enabled.

Yessss, I definitely use authentication.

mindsuru, 2023-11-23:
Okay good to know. I will have a look on it. But unfortunatly i‘m ill right now. It may take a bit.

viktak, 2023-11-23:
> Okay good to know. I will have a look on it. But unfortunatly i‘m ill right now. It may take a bit.

Get well soon, that's the priority!

mindsuru, 2023-11-24:
i think thats it. now both upload methods should work. regardless of whether you use the encrypted or non-encrypted method.
i opened a pull request for that change

viktak, 2023-11-24:
@mindsuru Yes, it is now working.
I just checked the version in the pull request, and at first it failed with a message that there was no username defined, but once I defined all the credentials the upload worked flawlessly.
Here are all the relevant additions in `platformio.ini` I had to make it work:
```
extra_scripts = 
    platformio_upload.py

build_flags =
    '-DELEGANTOTA_USE_ASYNC_WEBSERVER = 1'

lib_deps =
    ottowinter/ESPAsyncWebServer-esphome @ ^3.1.0
    esphome/AsyncTCP-esphome @ ^2.0.1
    ayushsharma82/ElegantOTA @ ^3.1.0

upload_protocol = custom
upload_url = http://w.x.y.z
username = username 
password = password 
```

viktak, 2023-11-24:
One more request, perhaps it is possible to rename the expected variables from this:
```
upload_url = http://w.x.y.z
username = username 
password = password 
```
to this:
```
custom_upload_url = http://w.x.y.z
custom_username = username 
custom_password = password 
```
This is to get rid of those pesky PlatformIO warning:
```
Warning! Ignore unknown configuration option `upload_url` in section [env:esp32]
Warning! Ignore unknown configuration option `username` in section [env:esp32]
Warning! Ignore unknown configuration option `password` in section [env:esp32]
```

mindsuru, 2023-11-24:
sure and thanks for this information

viktak, 2023-12-29:
I have been using the updated script for a couple of months now on various projects. Here are my observations:

1. Doing an update from the /update page works all the time, every time, be it the firmware or the spiffs.
2. From PlatformIO, uploading the code works perfectly all the time, every time.
3. However, uploading the spiffs from PlatformIO is a hit and miss for me, and I haven't figured out what the problem might be. I haven't even noticed a pattern, to be honest. Sometimes it works, sometimes it fails during upload, sometimes it goes fine to 100%, then it says:
```
Upload Progress: 100%|██████████| 1.50M/1.50M [00:07<00:00, 205kB/s]

Upload faild.
Server response: Failed to write chunked data to free space
```
or it complains about the magic byte (I can't remember the exact message...)
4. Some typos in the console: "faild" *3 and "successfull" *1

viktak, 2024-07-05:
I still have this issue.

viktak, 2024-12-12:
ping

closed 2025-01-13 by ayushsharma82