ayushsharma82/ElegantOTA #74: OTA not working on somewhat weaker WIFI strength

TPD95100, 2022-01-29:
As many others I discover that OTA ends with an error message when I do not have good WIFI signal strength. It stops with the message [HTTP ERROR] Bad Request in the update-Webpage.

I use it on an ESP32 and it runs a webserver with a small webpage to control a few switches and it has stable MQTT-Connection to Node-Red. So, generally, communication over the network is working reliable.

Looking at other comments, I feel that ELegantOTY is particularly demanding a very high network quality - As many other services work with much lower WIFI Signal strength (And I use many ESP32 at places with less than ideal signal strength without any problem except that I cant run OTA) I hope that there will be a way to correct this (maybe there is some error correction / checksum thing to da as I am sure that every means to avoid installing a corrupted binary is present :-) 

Other than that I am very happy with ElegentOTA

t-jones14, 2022-02-16:
Any type of OTA Update should only be done with a consistent WiFi signal strength. The reason it's probably failing is due to the upload being interrupted by said poor WiFi connection. The only way around this would be to constantly have the client-side scripting uploading and verifying with the ESP8266/32 each chunk that's uploaded and holding it in-state if a response isn't received and resuming once reconnected. This would add a considerable amount of code and also delay upload and write speed. It would be a resource-intensive task that these boards more than likely just don't have the capacity for.

Another way of looking at this is, for each new line of code added to ElegantOTA, is less space available for your sketch and upload(ing)ed OTA update since the base ElegantOTA library and clientside code takes up space. (Refer to this link)[https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#update-process-memory-view]

ayushsharma82, 2023-09-23:
Answered by @t-jones14 . Thanks for an informative answer!

closed 2023-09-23 by ayushsharma82