ayushsharma82/ElegantOTA #92: Info needed: how to upload from an App ?

Alisofin, 2022-09-20:
Hello,
I'd like to upload a new code to an ESP32 board that uses ElegantOTA, through a phone App I'm currently writing. I guess the App should send an HTML post to IP-address-of-ESP32/update with the new binary file as parameter ? Could you please confirm this and what are the parameters to be used ?
Many thanks in advance.

goncamateus, 2022-09-29:
Up! I'm facing the same problem, but I didn't find the parameter to change the file type.

Franck78, 2023-03-12:
Hi
Why would AsyncWebServer rewrite http(s) protocol ?

Launch **tcpdump** or **WireShark** on your workstation, then study the TCP sessions between the Esp32 and your browser.

Then use your preferred scripted Perl/Python/Php tool to do the same. Also read the HTTP libraries documentation to not reinvent the wheel !

ayushsharma82, 2023-10-17:
You'll have to decode the HTTP API used within ElegantOTA. Unfortunately this is not documented.

closed 2023-10-17 by ayushsharma82