ayushsharma82/ElegantOTA #100: /update doesn't work on a ESP32cam board

alport, 2022-11-18:
I am trying to upload the LED .bin file via the OTA on a ESP32-CAM board using: 
https://randomnerdtutorials.com/esp32-ota-over-the-air-arduino/

Problem is that when I try to browse to the IP: 192.168.0.114/update after a little delay, I just get a blank screen – ie no web page.
However, when I browse to 192.168.0.114 I get the expected: “I am ESP32”.
Could be a problem unique to the ESP32cam board.
I have tried updating via OTA and /update using the same program on a standard ESP32 - and it works with no problem.
Can anyone point me in the right direction?

KeithHanson, 2023-01-30:
Hm. Can confirm, I'm having issues as well. I've managed to capture this log if it seems useful to anyone:

```
update:24          POST http://192.168.1.36/update net::ERR_CONNECTION_RESET
(anonymous) @ update:24
Promise.then (async)
uploadOTA @ update:24
re @ update:24
r @ update:24
Ko.i._wrapper @ update:24
```

After which, the device resets. The firmware is exported from Arduino 1.8.19 (sketch -> export as compiled binary) and is 1mb. Am I running out of space perhaps?

KeithHanson, 2023-01-30:
Digging into this, it looks like it's due to the partition scheme not having enough space for the incoming firmware. 

Haven't gotten any of the options in the Arduino IDE to work, though with ElegantOTA. Will report back if I've solved it.

KeithHanson, 2023-01-30:
Another user reporting similar feedback here: https://github.com/ayushsharma82/ElegantOTA/issues/89

KeithHanson, 2023-01-30:
I have tried so many different configurations now of OTA partitions. Default sketch for ElegantOTA works over many different versions of partition layouts. 

Once I add in my libraries, things fail:
```
#include "WiFi.h"
#include "esp_camera.h"
#include "esp_timer.h"
#include "img_converters.h"
#include "Arduino.h"
#include "soc/soc.h"           // Disable brownour problems
#include "soc/rtc_cntl_reg.h"  // Disable brownour problems
#include "driver/rtc_io.h"

#include <ArduinoWebsockets.h>

#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <AsyncElegantOTA.h>
```

Going to try and whittle down which library it is next.

KeithHanson, 2023-01-31:
Whelp. After wrestling with this all day, I just decided to abandon the current arduino project, start from scratch. 

In Arduino 1.8.19, using the github master zip downloads for all required libraries, I'm able to successfully:

1. Take a picture
2. Upload it over websocket to my server
3. Wait for a predetermined amount of time
4. Deep sleep for a predetermined amount of time

While waiting to sleep, the ElegantOTA works perfectly. 

My settings are obvious:

Board: Ai Thinker ESP32-CAM
Partition Scheme: Minimal SPIFFS

I'm pretty much at a loss as to what was happening before, but a fresh Arduino project fixed it for me.

sopek777, 2023-02-19:
Тоже столкнулся с такой проблемой на esp8266. Сразу после загрузки esp страница ip/update работает. Потом, через некоторое время не загружается. Может пройти час, а может и несколько дней беспрерывной работы esp чтобы вылезла эта проблема

De-Backer, 2023-07-21:
hello everyone, after having problems with it yesterday it seems to be the partitions.
this did work with "partitions = default.csv"

https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default.csv

PlatformIO.ini
`[env:esp32cam]
platform = espressif32
board = esp32cam
framework = arduino
monitor_speed = 115200
board_build.partitions = default.csv
lib_deps = 
    ayushsharma82/AsyncElegantOTA
    ottowinter/ESPAsyncWebServer-esphome`

by default this is huge_app.csv in platformio
see:
https://github.com/platformio/platform-espressif32/blob/master/boards/esp32cam.json

i hope this helps.

mathieucarbou, 2023-12-25:
This is already in the official Expressif doc. Whatever system is used for OTA (Arduino OTA, ElegantOTA or anything else), it requires that the partitions are correctly set. This is unrelated to ow ElegantOTA is working but is related to how the Upload feature works in Expressif.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/ota.html

> The OTA update mechanism allows a device to update itself based on data received while the normal firmware is running (for example, over Wi-Fi or Bluetooth.)
> 
>OTA requires configuring the [Partition Tables](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html) of the device with at least two OTA app slot partitions (i.e., ota_0 and ota_1) and an OTA Data Partition.

closed 2024-07-08 by github-actions[bot]