ayushsharma82/ElegantOTA #101: WT32-ETH01 - work via Ethernet port

Sergio-tix, 2023-01-08:
Failed to get it to work via ethernet on board WT32-ETH01. Lots of conflicts between libraries.
```
In file included from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6\libraries\WebServer\src/HTTP_Method.h:4,
                 from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6\libraries\WebServer\src/WebServer.h:30,
                 from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\libraries\WebServer_WT32_ETH01-main\src/WebServer_WT32_ETH01.h:84,
                 from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\ESP32_ElegantOTA_Demo\ESP32_ElegantOTA_Demo.ino:11:
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:95:6: error: 'HTTP_DELETE' conflicts with a previous declaration
   XX(0,  DELETE,      DELETE)       \
      ^
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX'
 #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\ESP32_ElegantOTA_Demo\ESP32_ElegantOTA_Demo.ino:5:
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:64:3: note: previous declaration 'WebRequestMethod HTTP_DELETE'
   HTTP_DELETE  = 0b00000100,
   ^~~~~~~~~~~
In file included from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6\libraries\WebServer\src/HTTP_Method.h:4,
                 from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6\libraries\WebServer\src/WebServer.h:30,
                 from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\libraries\WebServer_WT32_ETH01-main\src/WebServer_WT32_ETH01.h:84,
                 from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\ESP32_ElegantOTA_Demo\ESP32_ElegantOTA_Demo.ino:11:
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:96:6: error: 'HTTP_GET' conflicts with a previous declaration
   XX(1,  GET,         GET)          \
      ^
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX'
 #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\ESP32_ElegantOTA_Demo\ESP32_ElegantOTA_Demo.ino:5:
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:62:3: note: previous declaration 'WebRequestMethod HTTP_GET'
   HTTP_GET     = 0b00000001,
   ^~~~~~~~
In file included from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6\libraries\WebServer\src/HTTP_Method.h:4,
                 from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6\libraries\WebServer\src/WebServer.h:30,
                 from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\libraries\WebServer_WT32_ETH01-main\src/WebServer_WT32_ETH01.h:84,
                 from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\ESP32_ElegantOTA_Demo\ESP32_ElegantOTA_Demo.ino:11:
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:97:6: error: 'HTTP_HEAD' conflicts with a previous declaration
   XX(2,  HEAD,        HEAD)         \
      ^
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX'
 #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\ESP32_ElegantOTA_Demo\ESP32_ElegantOTA_Demo.ino:5:
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:67:3: note: previous declaration 'WebRequestMethod HTTP_HEAD'
   HTTP_HEAD    = 0b00100000,
   ^~~~~~~~~
In file included from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6\libraries\WebServer\src/HTTP_Method.h:4,
                 from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6\libraries\WebServer\src/WebServer.h:30,
                 from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\libraries\WebServer_WT32_ETH01-main\src/WebServer_WT32_ETH01.h:84,
                 from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\ESP32_ElegantOTA_Demo\ESP32_ElegantOTA_Demo.ino:11:
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:98:6: error: 'HTTP_POST' conflicts with a previous declaration
   XX(3,  POST,        POST)         \
      ^
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX'
 #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\ESP32_ElegantOTA_Demo\ESP32_ElegantOTA_Demo.ino:5:
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:63:3: note: previous declaration 'WebRequestMethod HTTP_POST'
   HTTP_POST    = 0b00000010,
   ^~~~~~~~~
[... 30 lines cut ...]
In file included from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\ESP32_ElegantOTA_Demo\ESP32_ElegantOTA_Demo.ino:5:
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:68:3: note: previous declaration 'WebRequestMethod HTTP_OPTIONS'
   HTTP_OPTIONS = 0b01000000,
   ^~~~~~~~~~~~
In file included from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6\libraries\WebServer\src/HTTP_Method.h:4,
                 from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6\libraries\WebServer\src/WebServer.h:30,
                 from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\libraries\WebServer_WT32_ETH01-main\src/WebServer_WT32_ETH01.h:84,
                 from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\ESP32_ElegantOTA_Demo\ESP32_ElegantOTA_Demo.ino:11:
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:128:6: error: 'HTTP_PATCH' conflicts with a previous declaration
   XX(28, PATCH,       PATCH)        \
      ^~
[... cut ...]
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\ESP32_ElegantOTA_Demo\ESP32_ElegantOTA_Demo.ino:5:
C:\3_Arduino\arduino-1.8.19__esp32__ETH01__OTA\portable\!_sketchbook\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:66:3: note: previous declaration 'WebRequestMethod HTTP_PATCH'
   HTTP_PATCH   = 0b00010000,
   ^~~~~~~~~~

exit status 1

```

johnmx, 2023-01-18:
I use ElegantOTA v2.2.6 with a my WT32-ETH01 boards without any problems.

Sergio-tix, 2023-01-19:
> I use ElegantOTA v2.2.6 with my WT32-ETH01 boards without any problems.

What libraries do you use?
Could you provide a demo sketch?

johnmx, 2023-01-19:
> > I use ElegantOTA v2.2.6 with my WT32-ETH01 boards without any problems.
> 
> What libraries do you use? Could you provide a demo sketch?

I use the following:
 - framework-arduinoespressif32 @ 3.10006.210326 (1.0.6)
 - tool-esptoolpy @ 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 @ 2.50200.97 (5.2.0)
 - AsyncElegantOTA @ 2.2.6
 - AsyncTCP @ 1.1.1
 - WebServer_WT32_ETH01 @ 1.4.1

Sergio-tix, 2023-01-19:
> * WebServer_WT32_ETH01 @ 1.4.1
Thank you. Found.
https://github.com/khoih-prog/WebServer_WT32_ETH01/blob/main/examples/MQTT_And_OTA_Ethernet/MQTT_And_OTA_Ethernet.ino

ayushsharma82, 2023-09-23:
Answered by @johnmx

closed 2023-09-23 by ayushsharma82