ayushsharma82/ElegantOTA #104: Connot update...

phamiltonsmith, 2023-03-24:
Ok. I am unable to update ESP sketch w EOTA...

- I uploaded the original EOTA DEMO example to my esp8266 (sparkfun thing dev) using Arduino IDE v 2.0.4. Elegant OTA V 2.2.9. My device and computer are about 2m distance from wifi AP
- I connect to IPaddress/update. I can see the EOTA page and have firmware selected. ID coincides w ESP's MAC. all good.
- i connect to root and see: Hi! I am ESP8266. all good
- Meanwhile i have made a slight change to the root handle: FROM: ...server.send(200, "text/plain", "Hi! I am ESP8266");... TO: ...server.send(200, "text/plain", "Hi! I am ESP8266...");  ,, note 3 periods at the end
- I save the new modified sketch, export compiled binary. First ensuring there are NO old binaries around...
- I go back to the ip/update page and slide the new .bin file onto the page
- I get " OTA Success " msg. I click the "back" button...
- I then go back to the root handle and nothing has changed. I see "Hi! I am ESP8266" when (I reckon) I should see "Hi! I am ESP8266..." << w the three periods at the end.
- I viewed root from several browsers. same result

I have tried this same process on several different devices and appear to get the same results...

So where is my error?

MUCH appreciated and thanks for the great work!
P

phamiltonsmith, 2023-03-24:
Forgot to mention: 
- I also tried uploading the .bin file for blink WITHOUT all the EOTA stuff... nada, it did to upload << this would ensure complete "destruction" of the EOTA code ensuring I was indeed looking at something different and new...
- Also tried upload blink from 2 additional different browsers... nada << just in case some plug-in, ad blocker, etc was affecting the upload...

P

ayushsharma82, 2023-04-23:
Hi @phamiltonsmith ,

It's a common issue with ESP devices. You need to reboot them once after flashing through serial for them to be OTA updated successfully.

I hope this helps.

closed 2023-04-23 by ayushsharma82