ayushsharma82/ElegantOTA #113: Erase Flash on upload.

Duyvest, 2023-07-18:
With ArduinoIDE there is an option to erase flash on upload.  This can sometimes fix errors and crashes and is good to for a fresh start if a lot of changes have been made.  

Is this possible to do this with ElegantOTA? 

Thanks

Daryavahus, 2023-09-10:
I think it's not possible because of OTA library, right now. It's  stored in flash. When We would move library to IRAM, we would format flash and copy  uploaded file. I see that firmware uploaded directly from IDE via USB has less unexpexted error and reboots.
Or let the Arduino IDE  creates bin file with max available size - 1024KB(free space is empty).

ayushsharma82, 2023-09-23:
It is not possible when doing an OTA update. The process is completely different to how 'flash' procedure is handled by ArduinoIDE.

closed 2023-09-23 by ayushsharma82