ayushsharma82/ElegantOTA #114: Elegant OTA by URL

MikeF5J, 2023-07-28:
Ayush,
The ElegantOTA works well.  I would like to try and simplify the call to the device and do the FW update through an auto created URL.

Can you assist by helping with such a URL construct?
e.g.
ip/update/ x??x "fw filename & path" + button submitted

By sending the URL to the browser the IP should be started and the fw file selected and added and start the OTA automatically.

The coffees I'll save I'll pass to you.

Thank you
Mike

robertobuti, 2023-08-10:
Hello,
me too, i like same. OTA updated version working with a cloud storage coudl be a great idea.

awong1900, 2023-11-07:
I like this idea. I mange firmware on the S3, so if can from link to ota, it is fantastic.

ayushsharma82, 2023-11-22:
Not planned for ElegantOTA. The logic to do that is completely different from what ElegantOTA was originally designed for; which is  - a simple OTA webpage.

closed 2023-11-22 by ayushsharma82