ayushsharma82/ElegantOTA #121: [v3] add control over auth type

mathieucarbou, 2023-09-21:
For my need, right now I have to modify all the occurrences to: `request->requestAuthentication(NULL, false);`

Would be better to have a callback handle that instead of your code because auth type is application specific and using ElegantOTA right now in the context of an app with a different auth type than digest makes it request the password over and over.

ayushsharma82, 2023-09-23:
I actually don't know much other than HTTP basic authentication on ESP, can you describe more on what else is possible? 

*Note: It should be possible to do on both webservers and all platforms ( added RP2040 - Pico W recently ).*

mathieucarbou, 2023-09-23:
The problem in your OSS projects is that you are letting the http server handle how it should respond in case auth failed. This is wrong because the way to respond is highly app specific. The default implementation implied some default value in the `request->requestAuthentication();`. The problem is that someone would want to configure the realm name, others would want to control the used of basic vs digest auth.

Also, another use case (which I am using) in my project, and some other projects too is that the default answer is not suitable for an API call. When doing an API call, the suitable answer is a 401 error code, without the WWW-Authenticate header otherwise ajax calls would trigger a popup dialog instead of letting the UI handle the 401 error and present the login screen.

You can see such example here, were API calls (ajax) are using the `X-Requested-With`, especially to prevent such popops from the browser and handle the 401 error code. 

https://github.com/tbnobody/OpenDTU/blob/d419fd479461a933ffa13420e5448e0d45cec4b6/src/WebApi.cpp#L67-L83

In my project, here is my code to check for auth:

```c
    bool authenticated(AsyncWebServerRequest *request)
    {
      if (!_auth || request->authenticate(_username.c_str(), _password.c_str()))
        return true;
      AsyncWebServerResponse *r = request->beginResponse(401);
      // WebAPI should set the X-Requested-With to prevent browser internal auth dialogs
      if (!request->hasHeader("X-Requested-With"))
        r->addHeader("WWW-Authenticate", "Basic realm=\"Login Required\"");
      request->send(r);
      return false;
    }
```

What I really don't want is to call `requestAuthentication`()` because the default is not Basic but digest and it cannot handle ajax calls.

I don't want to use digest in order to simplify API calls on my app.

So in conclusion, I think you should allow the user to add a callback and handle itself the response of a failed auth.

mathieucarbou, 2023-09-23:
Bye the way, thanks a lot for all your OSS projects. They really have a nice UI. I bought ESP-Dash Pro recently, will use it in a project. I am also using this one and ESPConnect. Though, I must say, I had to do a few updates of the code ;-)

ayushsharma82, 2023-09-23:
Great to know that my OSS projects were of good use! ESPConnect was a "free-time-ware" where I just wanted a quick WiFiManager type library haha, I'll give it some more time in future. 

I'm thinking of your above suggestion, mind opening a PR where both strategies can co-exist?

mathieucarbou, 2023-09-23:
As soon as I have time I will, but the problem is that this one is not a 5-10 min change because the auth process is different depending on the web server impls so the callback signatures and also default behaviour have to be different too.

mathieucarbou, 2023-09-23:
Note that right now I am using the free version in my project, but If you would consider giving the pro version to contributors I would be glad to spend more time on that and also send further improvements I do along the road :-)

ayushsharma82, 2023-09-23:
> I would but the problem is that this one is not a 5-10 min change because the auth process is different depending on the web server impls

I was thinking the same thing but I've got people ( including you ) who use ElegantOTA for the API too. So it would be an interesting implementation.

> Note that right now I am using the free version in my project, but If you would consider giving the pro version to contributors

I do hand off 50% discount coupon to serious contributors. Contributors can send me a mail at ayush@softt.io to claim it. ( don't forget to mention your github username )

mathieucarbou, 2023-09-25:
I thought about the use case again... In my app, I can switch everything to digest and answer back to api calls with just a 401 like that

```cpp
    bool authenticated(AsyncWebServerRequest *request)
    {
      if (!_auth || request->authenticate(_username.c_str(), _password.c_str()))
        return true;
      if (request->hasHeader("X-Requested-With"))
        request->send(401);
      else
        request->requestAuthentication();
      return false;
    }
```

`requestAuthentication()` is quite dangerous to use because depending on the web serevr impl it does different things: either ask an auth with Basic, or Digect and the method has a different signature also.

ayushsharma82, 2023-11-22:
I'll mark it as not planned because it's a very niche use-case.

closed 2023-11-22 by ayushsharma82