ayushsharma82/ElegantOTA #127: doesn't restart after update

shrakanis, 2023-09-24:
doesn't restart after update is finished
wt32-eth01

simonttp78, 2023-09-25:
did you add ElegantOTA.loop() in your main loop()? That should solve your problem. Optionally you can add 2s delay and restart command in ElegantOTA.onEnd callback function. It should works fine too.

ayushsharma82, 2023-09-25:
I also suspect that it's due to not adding `ElegantOTA.loop()` in your `loop` block. It is responsible for rebooting your device.

> Optionally you can add 2s delay and restart command in ElegantOTA.onEnd callback function

@simonttp78  A 2 second delay is not needed anymore by user as library itself uses an internal 2 second timer to reboot automatically after successful OTA update.

shrakanis, 2023-09-25:
yes there is no ElegantOTA.loop(), but before update, worked ok
I'll try that, thanks

ayushsharma82, 2023-09-25:
Yea, as V3 was a major release, things have changed a bit. I’ll 100%
recommend reading docs again whenever there is a major release in library.

Anyways, I’m pretty sure adding that line will fix the issue. 😅 You may
close the issue after verifying.

closed 2023-10-02 by shrakanis