ayushsharma82/ElegantOTA #128: Errors compiling "AsyncDemo" on ESP32

Arubinu, 2023-09-24:
I was on AsyncElegantOTA which works perfectly but does not have a callback allowing you to know the progress.
When I went to the repository for it, I discovered that it was no longer maintained and that I had to switch to ElegantOTA.

After having conflicts during compilation I wanted to try the `AsyncDemo.ino` example, but the errors are exactly the same... Do you have any idea where I could have gone wrong?

I use the following repositories (as found on this repository in `.github/workflows/ci.yml`):
 - https://github.com/ayushsharma82/ElegantOTA.git
 - https://github.com/me-no-dev/AsyncTCP.git
 - https://github.com/me-no-dev/ESPAsyncWebServer.git

```
Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (6.3.2) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ElegantOTA @ 3.0.0+sha.8763d72
|-- AsyncTCP @ 1.1.1+sha.ca8ac5f
|-- ESP Async WebServer @ 1.2.3+sha.f71e3d4
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\esp32doit-devkit-v1\src\main.cpp.o
Building .pio\build\esp32doit-devkit-v1\bootloader.bin
Generating partitions .pio\build\esp32doit-devkit-v1\partitions.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio\build\esp32doit-devkit-v1\lib469\AsyncTCP@src-7fb2940bccb78b8d2de6915ae328b7fc\AsyncTCP.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\liba46\FS\FS.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\liba46\FS\vfs_api.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib09e\WiFi\WiFi.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib09e\WiFi\WiFiAP.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib09e\WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib09e\WiFi\WiFiGeneric.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib09e\WiFi\WiFiMulti.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib09e\WiFi\WiFiSTA.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib09e\WiFi\WiFiScan.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib09e\WiFi\WiFiServer.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib09e\WiFi\WiFiUdp.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib8c4\ESP Async WebServer\AsyncEventSource.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib8c4\ESP Async WebServer\AsyncWebSocket.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\lib8c4\ESP Async WebServer\SPIFFSEditor.cpp.o
In file included from C:/Users/pyrof/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/HTTP_Method.h:4,
                 from C:/Users/pyrof/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:30,
                 from .pio/libdeps/esp32doit-devkit-v1/ElegantOTA/src/ElegantOTA.h:73,
                 from src/main.cpp:12:
C:/Users/pyrof/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:95:6: error: 'HTTP_DELETE' conflicts with a previous declaration
   XX(0,  DELETE,      DELETE)       \
      ^
C:/Users/pyrof/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX'
 #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
C:/Users/pyrof/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from src/main.cpp:11:
.pio/libdeps/esp32doit-devkit-v1/ESP Async WebServer/src/ESPAsyncWebServer.h:64:3: note: previous declaration 'WebRequestMethod HTTP_DELETE'
   HTTP_DELETE  = 0b00000100,
   ^~~~~~~~~~~
[... 88 lines cut ...]
                 from src/main.cpp:12:
C:/Users/pyrof/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:128:6: error: 'HTTP_PATCH' conflicts with a previous declaration
   XX(28, PATCH,       PATCH)        \
      ^~
C:/Users/pyrof/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX'
 #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
C:/Users/pyrof/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from src/main.cpp:11:
.pio/libdeps/esp32doit-devkit-v1/ESP Async WebServer/src/ESPAsyncWebServer.h:66:3: note: previous declaration 'WebRequestMethod HTTP_PATCH'
   HTTP_PATCH   = 0b00010000,
   ^~~~~~~~~~
src/main.cpp: In function 'void setup()':
src/main.cpp:66:27: error: no matching function for call to 'ElegantOTAClass::begin(AsyncWebServer*)'
   ElegantOTA.begin(&server);    // Start ElegantOTA
                           ^
In file included from src/main.cpp:12:
.pio/libdeps/esp32doit-devkit-v1/ElegantOTA/src/ElegantOTA.h:107:10: note: candidate: 'void ElegantOTAClass::begin(WebServer*, const char*, const char*)'
     void begin(ELEGANTOTA_WEBSERVER *server, const char * username = "", const char * password = "");
          ^~~~~
.pio/libdeps/esp32doit-devkit-v1/ElegantOTA/src/ElegantOTA.h:107:10: note:   no known conversion for argument 1 from 'AsyncWebServer*' to 'WebServer*'
*** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1
```

ayushsharma82, 2023-09-24:
> Compiling .***@***.***\AsyncTCP.cpp.o

Errors suggest that you have both Webserver and AsyncWebServer included in
your code. Remove the “Webserver.h” incase you are using AsyncWebServer.

Arubinu, 2023-09-24:
I exclusively use the code from the example file given by this library, nothing more.

ayushsharma82, 2023-09-24:
😄 Always read docs: https://docs.elegantota.pro/async-mode/ 

Tried compiling on PIO. It works after you enable async mode.

ayushsharma82, 2023-09-24:
I'll make it more clear in the AsyncDemo example though. I hope this fixes it.

Arubinu, 2023-09-24:
Ah damn, sorry, but why not include `#define ELEGANTOTA_USE_ASYNC_WEBSERVER 1` in the example so that it is directly operational? I saw that by positioning it before putting `#include <ElegantOTA.h>` is supported by your library :)

On the other hand, I have only one problem that happens during compilation (which does not occur when I delete the `begin` line):
```
PLATFORM: Espressif 32 (6.3.2) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ElegantOTA @ 3.0.0+sha.8763d72
|-- ESP Async WebServer @ 1.2.3+sha.f71e3d4
|-- AsyncTCP @ 1.1.1
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio\build\esp32doit-devkit-v1\src\main.cpp.o
Linking .pio\build\esp32doit-devkit-v1\firmware.elf
c:/users/pyrof/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32doit-devkit-v1\src\main.cpp.o:(.literal._Z5setupv+0x68): undefined reference to `ElegantOTAClass::begin(AsyncWebServer*, char const*, char const*)'
c:/users/pyrof/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32doit-devkit-v1\src\main.cpp.o: in function `setup()':
C:\Users\pyrof\Documents\PlatformIO\Projects\OTATest/src/main.cpp:67: undefined reference to `ElegantOTAClass::begin(AsyncWebServer*, char const*, char const*)'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32doit-devkit-v1\firmware.elf] Error 1
```

ayushsharma82, 2023-09-24:
It will not compile correctly by that way.

The best way is to just follow what’s written in the docs which works for
both ArduinoIDE and PlatformIO.

Arubinu, 2023-09-24:
Indeed, by modifying directly in the library it is functional.

On the other hand, it's not great when you want to share a PlatformIO project, since they will also have to modify the library.

Thanks for the help anyway.

Arubinu, 2023-09-24:
Besides, the pro version is cool and must have required a lot of work.
When I have the chance, I will support your work by purchasing this version <3

ayushsharma82, 2023-09-24:
You can try to use build_flags in PIO by setting
“-DELEGANTOTA_USE_ASYNC_WEBSERVER=1”  but most people don’t know it, docs
has the most simple way to do it.

Anyways, I’m glad the issue is resolved.

Arubinu, 2023-09-24:
It works wonderfully, thank you very much :)

closed 2023-09-24 by Arubinu

MaffooClock, 2023-10-09:
> Indeed, by modifying directly in the library it is functional.
> 
> On the other hand, it's not great when you want to share a PlatformIO project, since they will also have to modify the library.

This is my sentiment exactly -- requiring the end user of a library to _modify_ the library seems like bad form.  Updates will overwrite the changes, as will cleaning the build, so the user will have to remember to re-apply the change every time.  👎 

Instead, why not just check whether [`_ESPAsyncWebServer_H_`](https://github.com/me-no-dev/ESPAsyncWebServer/blob/f71e3d427b5be9791a8a2c93cf8079792c3a9a26/src/ESPAsyncWebServer.h#L22C30-L22C30) is defined as a way to determine which one to use?

ayushsharma82, 2023-10-19:
@MaffooClock  I've added another simple solution in my comment above if you didn't go through the whole conversation. You can simply use build_flags of PlatformIO to achieve the same without modifying the library.

ayushsharma82, 2023-10-19:
Updated docs: https://docs.elegantota.pro/async-mode/#for-platformio