ayushsharma82/ElegantOTA #129: ElegantOTA.cpp:73:71: error: 'FS_end' was not declared in this scope;

VaAndCob, 2023-09-25:
I installed library both on PC (windows) and Apple Mac(intel5)   both machine are Arduino 2.2.1
- Arduino on PC (windows10) compiled and run fine.
- Arduino on macbook throw error messages:
/Volumes/sdcard64gb/Dropbox/Arduino/Sketch/libraries/ElegantOTA/src/ElegantOTA.cpp: In lambda function:
/Volumes/sdcard64gb/Dropbox/Arduino/Sketch/libraries/ElegantOTA/src/ElegantOTA.cpp:73:71: error: 'FS_end' was not declared in this scope; did you mean 'va_end'?
   73 |         uint32_t update_size = mode == OTA_MODE_FILESYSTEM ? ((size_t)FS_end - (size_t)FS_start) : ((ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000);
      |                                                                       ^~~~~~
      |                                                                       va_end
/Volumes/sdcard64gb/Dropbox/Arduino/Sketch/libraries/ElegantOTA/src/ElegantOTA.cpp:73:88: error: 'FS_start' was not declared in this scope; did you mean 'va_start'?
   73 |         uint32_t update_size = mode == OTA_MODE_FILESYSTEM ? ((size_t)FS_end - (size_t)FS_start) : ((ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000);
      |                                                                                        ^~~~~~~~
      |                                                                                        va_start

ayushsharma82, 2023-09-25:
Check if both PC and macbook have the same board version installed?

VaAndCob, 2023-09-25:
it's exactly the same ESP8266 module,  (Generic ESP8266)  but Arduino IDE was unable to compile due to error messages above.  (both PC and Macbook use  library and sketch code from the same resource (dropbox) .
I wonder why PC was successful compile but not macbook.   Are "FS_end" and "FS_start" not declare error come from different "FS.h"  ?

VaAndCob, 2023-09-25:
it's the latest version.  ESP8266 Board(3.0.2).  No update both machine lately.

ayushsharma82, 2023-09-25:
@VaAndCob  It's an issue in ESP8266 Board 3.0.2 version. I compiled the example successfully with 3.1.2 (latest) on my MacBook (M1)

VaAndCob, 2023-09-25:
I manually copied and replace arduino core library file FS.h and FS.cpp from github [ESP8266 arduino core](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/FS.h)
but it gave the same error messages.

VaAndCob, 2023-09-25:
> I manually copied and replace arduino core library file FS.h and FS.cpp from github [ESP8266 arduino core](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/FS.h) but it gave the same error messages

ayushsharma82, 2023-09-25:
> I manually copied and replace arduino core library file FS.h and FS.cpp from github [ESP8266 arduino core](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/FS.h) but it gave the same error messages.

Consider updating to 3.1.2 release of ESP8266 board. It does not compile with 3.0.2

VaAndCob, 2023-09-25:
> > I manually copied and replace arduino core library file FS.h and FS.cpp from github [ESP8266 arduino core](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/FS.h) but it gave the same error messages.
> 
> Consider updating to 3.1.2 release of ESP8266 board. It does not compile with 3.0.2

oh ok,  I must manually update to 3.1.2  Thank you a  lot.

ayushsharma82, 2023-09-25:
@VaAndCob As this is a new release, I will need a small feedback. Once you have tried ElegantOTA, let me know if it updated your ESP8266 without any hiccups or not.

VaAndCob, 2023-09-25:
[image]

I confirm it works like a charm.

ayushsharma82, 2023-09-25:
Awesome! Thanks for the feedback.

closed 2023-09-25 by ayushsharma82