ayushsharma82/ElegantOTA #130: Callback onEnd reports "failure" while WebUI reports success

simonttp78, 2023-09-25:
After reboot I can see that new firmware is reported via console, therefore print in callback onEnd is reporting false feedback.

Environment: VSCode for Arm64 (M1 Mac), PlatformIO 6.1.11, Espressif 32 version 6.3.2 (framework-arduinoespressif32 ver. 2.0.9), ESP32 LoLin

console:
[ElegantOTA]  OTA Mode: Firmware
[ElegantOTA]  MD5: 441018525208457705bf09a8ee3c1093
Firmware update via WebUI initiated. <-- callback function onStart
[ 28616][D][Updater.cpp:133] begin(): OTA Partition: app0
Update failed. <-- callback function onEnd
[ElegantOTA]  Rebooting...

ayushsharma82, 2023-09-25:
This has been fixed with v3.1.0 which was released just a couple of hours
ago.

simonttp78, 2023-09-25:
Works fine with 3.1.0. Thank you.

closed 2023-09-25 by simonttp78