ayushsharma82/ElegantOTA #133: ElegantOTA in AP mode

savergiggio, 2023-09-29:
Hello guys, i'm trying to let work the ESP32 in AP mode, bit i have some issues, for example during the upload of the code the ESP reboot.

Anyone tried this type of configuration or have the same problem?

Thank you in advanced!

washoegary, 2023-10-04:
I to am trying to use the softAP mode without success. The progress bar completes to 100% and then displays a 'MD5 Check Failed' message. On the serial monitor, the OTA progress bytes never end up matching the final bytes. I don't get a reboot, and the existing code continues normally. I have tried stopping all application code in the main loop  (except ElegantOTA.loop();)  during updates with no effect.

washoegary, 2023-10-05:
My issue was was caused by not using Arduino IDE 2.0. It's working just fine now. I can use my android phone in the field to update firmware on ESP32's installed on top of 150' water towers and miles from any WiFi station!  softAP defaults to an IP of 192.168.4.1 but can be explicitly set to whatever you like. Now, if anyone knows how to OTA Update via a Cat-M1 cellular connection with a 9600 baud serial link between the cellular modem and esp32 let me know.

savergiggio, 2023-10-05:
> My issue was was caused by not using Arduino IDE 2.0. It's working just fine now. I can use my android phone in the field to update firmware on ESP32's installed on top of 150' water towers and [...]

I use VS Code, i'm using wifi with bluetooth, but i tried to disable bluetooth when updating OTA i way to not interrupt the update process, but the problem still relain. Now i discovered that maybe the problem could be the use of the DAC 2 which could cause  noises to the wifi connection and so interrupt the update.

ayushsharma82, 2023-10-17:
Hi @savergiggio , Please use the `onStart` and `onEnd` callbacks provided by ElegantOTA to shut down any extra tasks before doing OTA updates. This is necessary for smooth and crash free experience.

closed 2023-11-22 by ayushsharma82