ayushsharma82/ElegantOTA #135: Compatibility issues with AsyncWebServer installed thru library manager

Augustinas01, 2023-10-11:
When i include <AsyncElegantOTA.h> then it won't compile and I get 
[image]
Here are my includes
[image]
I'm using Arduino IDE, libraries installed thru library manager

Augustinas01, 2023-10-11:
I been able to compile it successfully by pulling ESPAsyncWebServer from git, then it gave me errors not finding methods in ElegantOTA, so I changed lib by deleting all chip checking and declaring ELEGANTOTA_USE_ASYNC_WEBSERVER 1 inside lib, now let's hope it's actually working.

ayushsharma82, 2023-10-11:
I noticed your included library is different than what ElegantOTA uses.
Please use the official ESPAsyncWebserver library instead of forks.

closed 2023-10-17 by ayushsharma82