ayushsharma82/ElegantOTA #136: OTA not working on ESP8285

sll1999, 2023-10-15:
Hi,

I tried to do a firmware update with ElegantOTA on a 1M ESP8285 and installed the demo code.
Flashing seems to work, but after this the 8285 is not able to boot the code:

--------------------------------------
OTA update started!
OTA Progress Current: 418 bytes, Final: 328969 bytes
OTA Progress Current: 62594 bytes, Final: 328969 bytes
OTA Progress Current: 135490 bytes, Final: 328969 bytes
OTA Progress Current: 215354 bytes, Final: 328969 bytes
OTA Progress Current: 289322 bytes, Final: 328969 bytes
OTA update finished successfully!
⸮@*r*rA(⸮⸮K5⸮P⸮⸮XKA⸮~⸮⸮Ѫ⸮KWI⸮⸮⸮I*E,⸮⸮⸮⸮⸮P⸮B!⸮~A⸮!⸮⸮n⸮*E⸮Hh⸮⸮A,P'!⸮⸮E,R⸮⸮⸮⸮I@vA.⸮⸮⸮(⸮⸮⸮|⸮	8x⸮vA,PZ
⸮vA,PZ
G,9V⸮⸮⸮⸮⸮⸮l *-⸮@*rjrA(⸮⸮S⸮u⸮P⸮v⸮X⸮(⸮⸮Pyju詅⸮8⸮t⸮⸮*N⸮~NP⸮T⸮⸮,)⸮vA-5⸮!⸮⸮A⸮!⸮⸮n⸮*E⸮⸮⸮InA,⸮%!⸮⸮E,R⸮Y⸮5⸮!u|⸮	
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0xc5
load 0x3c380028, len 8, room 8 
tail 0
chksum 0xa1
csum 0xa1
csum err
ets_main.c
--------------------------------------

Flash size is: FS:64kb OTA:470kb
Code size is: 323Kb
OTA size is: 323Kb

I also tried the same code on a ESP8266 board and it works flawlessly, even with the same flash layout.
Any clues why it doesn't work?

ayushsharma82, 2023-10-17:
No idea. Let me know if you find any pattern.

closed 2023-10-17 by ayushsharma82

zyssai, 2024-09-29:
Hi, someone managed to get it working?
I have an ESP-03 with the same error. Update is done, but code does not start anymore.

```
Update Received: ESP-03_OTA_minimal.bin

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v0005d350
@cp:B
```

Here is while flashing same file with Arduino IDE:

```
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ********************
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.1s
Compressed 381776 bytes to 278142...
Writing at 0x00000000... (5 %)
Writing at 0x00004000... (11 %)
Writing at 0x00008000... (17 %)
Writing at 0x0000c000... (23 %)
Writing at 0x00010000... (29 %)
Writing at 0x00014000... (35 %)
Writing at 0x00018000... (41 %)
Writing at 0x0001c000... (47 %)
Writing at 0x00020000... (52 %)
Writing at 0x00024000... (58 %)
Writing at 0x00028000... (64 %)
Writing at 0x0002c000... (70 %)
Writing at 0x00030000... (76 %)
Writing at 0x00034000... (82 %)
Writing at 0x00038000... (88 %)
Writing at 0x0003c000... (94 %)
Writing at 0x00040000... (100 %)
Wrote 381776 bytes (278142 compressed) at 0x00000000 in 24.4 seconds (effective 124.9 kbit/s)...
Hash of data verified.
```

Any help appreciated.