ayushsharma82/ElegantOTA #139: on Examples AsyncDemo error

anwfeb, 2023-10-25:
AsyncDemo:88:20: error: cannot convert 'AsyncWebServer*' to 'ESP8266WebServer*' {aka 'esp8266webserver::ESP8266WebServerTemplate<WiFiServer>*'}
   88 |   ElegantOTA.begin(&server);    // Start ElegantOTA
      |                                     ^~~~~~~
      |                                     |
      |                                    AsyncWebServer*

ScruffR, 2023-10-29:
I got a similar issue with ESP32
```tet
error: no matching function for call to 'ElegantOTAClass::begin(AsyncWebServer*)'
   ElegantOTA.begin(&server);

note: candidate: 'void ElegantOTAClass::begin(WebServer*, const char*, const char*)'
     void begin(ELEGANTOTA_WEBSERVER *server, const char * username = "", const char * password = "");

note:   no known conversion for argument 1 from 'AsyncWebServer*' to 'WebServer*'
```

bmgmo-iscteiulpt, 2023-11-01:
I got the exact same error, cant figure out how to fix it :/

ayushsharma82, 2023-11-01:
Did you guys enable Async Mode as per docs before compiling the example?

bmgmo-iscteiulpt, 2023-11-01:
Yes, but same error

ayushsharma82, 2023-11-01:
Weird, CI gave it a pass. I'll need the following details to further diagnose it:

1. Are you using Arduino IDE or PlatformIO?
2. Which MCU are you using?
3. Board/Platform version of your microcontroller

bmgmo-iscteiulpt, 2023-11-01:
Arduino IDE
Its an Expressif ESP32-s2 wrover
The version I'm not sure how to get it 

[image]

bmgmo-iscteiulpt, 2023-11-01:
When the Async mode is set on the library file, it gives the following errors

[image]

ayushsharma82, 2023-11-01:
@bmgmo-iscteiulpt This is a different error compared to what  this issue was initially created for.

You can fix it by downgrading to a stable "arduino-esp32" release in Board Manager. Currently you are using v3.0.0-alphav2 which is not stable.

bmgmo-iscteiulpt, 2023-11-01:
worked! didnt even noticed the alpha version when installed the boards .. thank you!

ScruffR, 2023-11-02:
For an older project had it enabled but forgot to do that again after a library update cycle overrode that 😳

ayushsharma82, 2023-11-02:
@ScruffR  Yeah that's inevitable with Arduino IDE. I'll suggest moving to PlatformIO [with this approach](https://docs.elegantota.pro/async-mode/#for-platformio) so that the async flag is persisted within the project itself. One time learning effort, PIO offers a better experience in terms of compile time etc once you are familiar with it.

ayushsharma82, 2023-11-02:
@anwfeb Status update on your issue? If it was resolved, please close the issue.

Luckas007666, 2023-11-20:
@ayushsharma82 : Please also have a look at issue #141 . The issue itself is solved, but with a manual fix - maybe this can be added by you to the original ? https://github.com/ayushsharma82/ElegantOTA/issues/141#issuecomment-1808890201

closed 2024-01-07 by ayushsharma82

HARIS4820, 2024-06-27:
[image]