ayushsharma82/ElegantOTA #141: Error in AsyncDemo

Nikitanagar, 2023-11-09:
AsyncTCP.h and ESPAsyncWebServer.h files are not found. Can you let me know the solution?

Luckas007666, 2023-11-13:
when adding 
` "dependencies": { 
    "ottowinter/ESPAsyncWebServer-esphome": "^2.1.0",
    "ottowinter/ESPAsyncTCP-esphome": "1.2.3"
  },`

to library.json file of ElegantOTA, it compiles fine

Nikitanagar, 2023-11-20:
Thank you for the response Luckas, 
Yeah, This error has been solved after adding the above lines in library.json. But another error is found _**_argument of type "Asyncwebserver *"  is incompatible with parameter of type  "Webserver *"_**_

Luckas007666, 2023-11-20:
Did you enable Async Mode ?  https://docs.elegantota.pro/async-mode/#for-platformio

Nikitanagar, 2023-11-20:
Thank you it is working fine.

closed 2023-11-20 by Nikitanagar