ayushsharma82/ElegantOTA #142: File2Raw Utility Error

t-jones14, 2023-11-15:
File2Raw utility as outlined in ElegantOTAPro documentation is broken. Unable to access.

[image]

ayushsharma82, 2023-11-15:
File2Raw actually runs on my lab server and there is an internet connectivity issue going on in our area. It should be fixed in a day or two automatically.

t-jones14, 2023-11-15:
Thank you. Do you have alternate utility you would recommend by chance?

ayushsharma82, 2023-11-15:
File2Raw is the only recommended utility. You can keep trying again in a few minutes, it's just the fiber line which is acting up here.

ayushsharma82, 2023-11-22:
Connectivity issue was resolved a few days ago and File2Raw is online. Closing this issue.

closed 2023-11-22 by ayushsharma82