ayushsharma82/ElegantOTA #143: Demo.ino does not compile

JMWilton, 2023-11-17:
Using the ElegantOTA demo, unmodified.  Version 3.1.0 is installed.
Receive the following error:
...\Documents\Arduino\libraries\ElegantOTA\src\ElegantOTA.cpp:144:71: error: 'FS_end' was not declared in this scope
         uint32_t update_size = mode == OTA_MODE_FILESYSTEM ? ((size_t)FS_end - (size_t)FS_start) : ((ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000);

What am I missing?

mzaprudin, 2023-11-18:
I have the same issue

mzaprudin, 2023-11-18:
> Using the ElegantOTA demo, unmodified. Version 3.1.0 is installed. Receive the following error: ...\Documents\Arduino\libraries\ElegantOTA\src\ElegantOTA.cpp:144:71: error: 'FS_end' was not declared [...]

I've found solution
You should go to the ElegantOTA.cpp file and adjust FS_end and FS_star names to _FS_end and _FS_start (just add _ at the beginning)

JMWilton, 2023-11-18:
Thanks for a speedy reply.  I made the changes and tested.  It is now working.

mzaprudin, 2023-11-22:
> @JMWilton What microcontroller and board version (in Arduino IDE) are you using? This will help me diagnose a fix for the library.

I had the same issue on esp8266 with Platform IO
The problem is in ElegantOTA.cpp:144:71
Just need to fix FS_end and FS_start variable names, _ at the beginning is missed

ayushsharma82, 2023-11-22:
@mzaprudin , Yes I get it, but `FS_end` and `FS_start` variables are different for some versions of ESP8266 core. The library was compiled by CI successfully using ESP8266 platform.

I want to know if the variables were changed in latest ESP8266 platform/core or it's a user error where user is using a older version of ESP8266 platform/core.

mzaprudin, 2023-11-22:
> @mzaprudin , Yes I get it, but `FS_end` and `FS_start` variables are different for some versions of ESP8266 core. The library was compiled by CI successfully using ESP8266 platform.
>
> I want to know [...]

I use Platform 6.1.11 version

I found this core version information
\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\core_version.h
```
#define ARDUINO_ESP8266_GIT_VER   0xcf6ff4c4
#define ARDUINO_ESP8266_GIT_DESC  3.0.2
#define ARDUINO_ESP8266_VERSION   3.0.2

#define ARDUINO_ESP8266_MAJOR     3
#define ARDUINO_ESP8266_MINOR     0
#define ARDUINO_ESP8266_REVISION  2

#define ARDUINO_ESP8266_RELEASE   "3.0.2"
#define ARDUINO_ESP8266_RELEASE_3_0_2
```

JMWilton, 2023-11-22:
I am using NodeMCU 0.9 (ESP-12 Module).
Arduino IDE 2.1.1

mnvoronin, 2023-12-25:
> I found this core version information .platformio\packages\framework-arduinoespressif8266\cores\esp8266\core_version.h
>
> ```
> #define ARDUINO_ESP8266_GIT_VER   0xcf6ff4c4
> #define [...]

The latest Arduino core for ESP8266 is 3.1.2 and the error doesn't happen after updating to that version.

ayushsharma82, 2024-01-07:
For anyone facing this issue, As @mnvoronin mentioned - please update your ESP8266 / ESP32 platforms to latest release. 

ElegantOTA is always compiled and tested automatically using CI script on latest (stable) release of these platforms.

closed 2024-01-07 by ayushsharma82

JMWilton, 2024-01-07:
Thanks to Ayush Sharma.  I finally got it to build.  It would be great, for this an all projects, if the examples included the specific version levels being used.  But that would put a lot of work on developers that could be better spent on other tasks.  I was wondering if there was some way to include in the zip some portion of the build output to capture this information.  The goal would be NOT make a lot of work for the developer but provide configuration information for the end user to help in resolving these kinds of problems.  It is just a thought.  Thanks again to Ayush for his hard work.