ayushsharma82/ElegantOTA #148: ElegantOTA v 3.1.0 Error: FS_end  was not declared when setting ELEGANTOTA_USE_ASYNC_WEBSERVER = 1

dtill, 2023-12-07:
/Arduino/libraries/ElegantOTA/src/ElegantOTA.cpp:76:71: error: 'FS_end' was not declared in this scope; did you mean 'va_end'?
   76 |         uint32_t update_size = mode == OTA_MODE_FILESYSTEM ? ((size_t)FS_end - (size_t)FS_start) : ((ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000);

Board: Generic ESP8266

ElegantOTA v 3.1.0

DRN88, 2023-12-07:
Same error.

```
c:\Users\gorkhaan\Documents\Arduino\libraries\ElegantOTA\src\ElegantOTA.cpp: In lambda function:
c:\Users\gorkhaan\Documents\Arduino\libraries\ElegantOTA\src\ElegantOTA.cpp:144:71: error: 'va_end' was not declared in this scope
  144 |         uint32_t update_size = mode == OTA_MODE_FILESYSTEM ? ((size_t)va_end - (size_t)va_start) : ((ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000);
      |                                                                       ^~~~~~
c:\Users\gorkhaan\Documents\Arduino\libraries\ElegantOTA\src\ElegantOTA.cpp:144:88: error: 'va_start' was not declared in this scope
  144 |         uint32_t update_size = mode == OTA_MODE_FILESYSTEM ? ((size_t)va_end - (size_t)va_start) : ((ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000);
      |                                                                                        ^~~~~~~~

exit status 1

Compilation error: exit status 1
```

mnvoronin, 2023-12-25:
This error is caused by the outdated platform version in PlatformIO. Need to update it to the latest.

dandud100, 2024-01-01:
Had the same error in ArduinoIDE (esp8266 v3.0.2) the fix in #143 fixed it for me too!

ayushsharma82, 2024-01-07:
Ref: #143

closed 2024-01-07 by ayushsharma82