ayushsharma82/ElegantOTA #149: ESP32S3 cannot compile in any way due to conflicts

markeby, 2023-12-08:
Async Demo will not compile.  There is no way I could find to eliminate the conflicts.  I changed the order of include files many different ways.  This will not build!

C:\Users\markeby\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:95:6: error: 'HTTP_DELETE' conflicts with a previous declaration
   XX(0,  DELETE,      DELETE)       \
      ^
C:\Users\markeby\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX'
 #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
C:\Users\markeby\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14/tools/sdk/esp32s3/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from B:\Arduino Music\Synth32\Disp32\WebOTA.cpp:10:
C:\Users\markeby\Documents\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:64:3: note: previous declaration 'WebRequestMethod HTTP_DELETE'
   HTTP_DELETE  = 0b00000100,

JMWilton, 2023-12-26:
Ditto.  I am using Arduino IDE 2.2

mpboddie, 2023-12-29:
Same here, using platformIO in VSCode. I'm looking back in the code to confirm, but I think that ELEGANTOTA_USE_ASYNC_WEBSERVER should be set to 1 and that isn't happening.

JMWilton, 2023-12-29:
Arduino IDE 2.2.1
Additional Boards Manager: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
Board: NodeMCU 0.9 (ESP-12 Module)
Libraries:
AsyncTCP by dvarrel 1.1.4 installed
ESPAsyncTCP by dvarrel 1.2.4 installed
ElegantOTA by Ayush Sharma 3.1.0 installed

Error Messages:
c:\Users\Joseph Wilton\Documents\Arduino\libraries\ElegantOTA\src\ElegantOTA.cpp: In lambda function:
c:\Users\Joseph Wilton\Documents\Arduino\libraries\ElegantOTA\src\ElegantOTA.cpp:76:71: error: 'FS_end' was not declared in this scope
         uint32_t update_size = mode == OTA_MODE_FILESYSTEM ? ((size_t)FS_end - (size_t)FS_start) : ((ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000);
                                                                       ^
c:\Users\Joseph Wilton\Documents\Arduino\libraries\ElegantOTA\src\ElegantOTA.cpp:76:88: error: 'FS_start' was not declared in this scope
         uint32_t update_size = mode == OTA_MODE_FILESYSTEM ? ((size_t)FS_end - (size_t)FS_start) : ((ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000);
                                                                                        ^

exit status 1

Compilation error: exit status 1

mpboddie, 2023-12-29:
@markeby try the instructions here, https://docs.elegantota.pro/async-mode/

This should solve your issue.

ayushsharma82, 2024-01-07:
@markeby Did you get it working?

grantnorwood, 2024-01-08:
> @markeby try the instructions here, https://docs.elegantota.pro/async-mode/
> 
> This should solve your issue.

Following the async docs and using the example code does not compile.  Using the latest PlatformIO, building produces a ton of `#error "include FreeRTOS.h" must appear in source files before "..."` errors, beginning with this one:

```shell
In file included from .pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.h:29,
                 from src/main.cpp:29:
/Users/SomeUser/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/semphr.h:31:6: error: #error "include FreeRTOS.h" must appear in source files before "include semphr.h"
     #error "include FreeRTOS.h" must appear in source files before "include semphr.h"
```

I resolved the various `conflicts with a previous declaration` errors by ensuring that AsyncTCP wasn't listed twice in `/.pio/libdeps`.  Make sure that you're not listing `AsyncTCP` in your `lib_deps`, as ESPAsyncWebServer already brings that lib in for you.

```ini
lib_deps = 
  ayushsharma82/ElegantOTA@^3.1.0
  https://github.com/me-no-dev/ESPAsyncWebServer.git
  ; https://github.com/me-no-dev/AsyncTCP.git
build_flags=-DELEGANTOTA_USE_ASYNC_WEBSERVER=1
```

I also tried going into my `/.pio/libdeps/*/ElegantOTA` and updating the `#define ELEGANTOTA_USE_ASYNC_WEBSERVER 1` flag myself _(instead of using the `build_flags`)_, but without success.

mathieucarbou, 2024-03-03:
FYI here is what I am using on a big project using ElegantOTA, ESH-DASH and which works on ESP32, ESP32S, ESP32S2, ESP32S3, with and without ethernet:

`-D ELEGANTOTA_USE_ASYNC_WEBSERVER`

```
lib_deps =
  bblanchon/ArduinoJson @ 7.0.3
  esphome/AsyncTCP-esphome @ 2.1.3
  mathieucarbou/ESP Async WebServer @ 2.7.0
  https://github.com/ayushsharma82/ElegantOTA
```

closed 2024-07-07 by github-actions[bot]

birolgithub, 2024-08-21:
Hello
I tried using ElegantOTA. It conflicted with AsyncTCP. I uninstalled AsyncTCP. When I uninstalled it, it gave me a WiFiManager error. Don't we have a chance to use them together? thanks