ayushsharma82/ElegantOTA #153: Improvement request for README - show that "/update" is the url to use for ota

HowardsPlayPen, 2023-12-20:
Firstly, this is a great package and good to use - saving lots of effort in updating firmware.

My only pause and wtf moment - was when I had uploaded the demo code to my ESP32 and I could see the stock response on  '/' but I had no idea on how to actually connect to the ota update part as the documentation did not mention anywhere the actual url to use. I looked at the code and cycled through the visible urls of "/ota/start" and then on to "/update" which finally got me to the right page

Please please, can you pop a small comment in the README (and in the example source code) of the url to go to in order to see the ElegantOTA webpage

Otherwise, thanks for a great product.

LDprg, 2023-12-30:
Yeah this one! Took me about 30 min to find this out.

Maybe a way to change the path would also be nice (and if possible doc about it).

ayushsharma82, 2024-01-04:
Didn't realise. I'll add it in docs soon.

closed 2024-01-05 by HowardsPlayPen