ayushsharma82/ElegantOTA #156: Possibility to run ElegantOTA with PsychicHttp

T94T, 2023-12-30:
Is there an easy way to run `ElegantOTA` in combination with https://github.com/hoeken/PsychicHttp instead of `ESPAsyncWebServer`?

HowardsPlayPen, 2023-12-31:
I was just about to raise an issue to see if there was any desire (or backlash) against submitting a PR with a patch to allow a compile flag to incorporate PsychicHttp - as I have a working version that does this

Is there any other view on this (I ask because AsyncWebServer does not look to be actively supported and PsychicHttp looks to be more efficient)

T94T, 2023-12-31:
Sounds great!

We use websockets a lot and it seems to run much more stable with PsychicHttp than with ESPAsyncWebServer. ESPAsyncWebServer is not maintained for a longer time and it has many open issues/PRs. In our opinion it would be a great feature to enable PsychicHttp by default.

HowardsPlayPen, 2023-12-31:
I just put my updates into a fork of ElegantOTA in https://github.com/HowardsPlayPen/ElegantOTA and I shall send a PR later with the tweaks (once I triple check that it all looks good). I have included a simple example that shows the build flag to use in platform.ini

T94T, 2023-12-31:
Really cool! I just tested your fork, and it works great.

ayushsharma82, 2024-01-04:
@HowardsPlayPen Let me know when PR is ready, PsychicHttp looks great.

HowardsPlayPen, 2024-01-05:
I have just sent a PR - let me know if there is anything you don't like..

T94T, 2024-02-14:
Uploading with `elegantota_upload.py` throws an error:
`<Response [405]> Request method for this URI is not handled by server`
Is there a server listener missing?

HowardsPlayPen, 2024-02-18:
Hello - apologies, I am not familiar with that python script or what it does. Do you think this is down to the PsychicHttp code I integrated (does it work with the regular code base?). I have not watched to see if ElegantOTA has been updated to know if there are any updates that might conflict somehow.
I am happy to have a look if you gave more details

T94T, 2024-02-19:
I just wanted to write a short description of how we tested the upload script. It seems that we had an older version of `platformio_upload.py` called `elegantota_upload.py`. Updating the script solved our problem.

With `platformio_upload.py` you can upload the firmware directly from VSCode to the device via OTA.

Steps to do:

1. Take the current `platformio_upload.py` script from https://raw.githubusercontent.com/ayushsharma82/ElegantOTA/master/platformio_upload.py and place it in the project's root folder beside `platformoi.ini`
2. Add the following configuration to the board's `[env]` section `platformio.ini`:
```
extra_scripts = platformio_upload.py
upload_protocol = custom
custom_upload_url = http://DEVICEADDRESS/update
```
4. Build and upload the project in VSCode (PlatformIO)

VSCode terminal output:
```
CURRENT: upload_protocol = custom
Uploading .pio/build/esp32-wroom-OTA/firmware.bin
Serverconfiguration: autentication not needed.

Upload Progress:   0%|          | 0.00/927k [00:00<?, ?B/s]
Upload Progress:  17%|█▋        | 160k/927k [00:00<00:01, 697kB/s]
Upload Progress:  26%|██▌       | 240k/927k [00:00<00:02, 255kB/s]
Upload Progress:  31%|███       | 288k/927k [00:01<00:03, 196kB/s]
Upload Progress:  35%|███▍      | 320k/927k [00:01<00:03, 194kB/s]
Upload Progress:  38%|███▊      | 352k/927k [00:01<00:03, 148kB/s]
Upload Progress:  41%|████▏     | 384k/927k [00:01<00:03, 163kB/s]
Upload Progress:  45%|████▍     | 416k/927k [00:02<00:03, 132kB/s]
Upload Progress:  48%|████▊     | 448k/927k [00:02<00:03, 146kB/s]
Upload Progress:  52%|█████▏    | 480k/927k [00:02<00:03, 116kB/s]
Upload Progress: 100%|██████████| 927k/927k [00:08<00:00, 116kB/s]

Upload successful.
Server response: OK
```

@ayushsharma82 could you please merge https://github.com/ayushsharma82/ElegantOTA/pull/159 from @HowardsPlayPen?

T94T, 2024-02-28:
@ayushsharma82 👍

closed 2024-02-28 by T94T