ayushsharma82/ElegantOTA #158: ElegantOTA Async Demo Example will not compile

HECSmith, 2024-01-04:
NodeMCU8266 12E board selected.  Otherwise the code is copied from the example.  Arduino 2.2.1 produces the following error message(s)
In file included from e:\H or P\Documents\Arduino\libraries\ElegantOTA\src/ElegantOTA.h:57,
                 from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:34:
C:\Users\H or P\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:47:19: error: 'HTTP_ANY' conflicts with a previous declaration
   47 | enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
      |                   ^~~~~~~~
In file included from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:33:
e:\H or P\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:69:3: note: previous declaration 'WebRequestMethod HTTP_ANY'
   69 |   HTTP_ANY     = 0b01111111,
      |   ^~~~~~~~
In file included from e:\H or P\Documents\Arduino\libraries\ElegantOTA\src/ElegantOTA.h:57,
                 from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:34:
C:\Users\H or P\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:47:29: error: 'HTTP_GET' conflicts with a previous declaration
   47 | enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
      |                             ^~~~~~~~
In file included from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:33:
e:\H or P\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:62:3: note: previous declaration 'WebRequestMethod HTTP_GET'
   62 |   HTTP_GET     = 0b00000001,
      |   ^~~~~~~~
In file included from e:\H or P\Documents\Arduino\libraries\ElegantOTA\src/ElegantOTA.h:57,
                 from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:34:
C:\Users\H or P\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:47:39: error: 'HTTP_HEAD' conflicts with a previous declaration
   47 | enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
      |                                       ^~~~~~~~~
In file included from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:33:
e:\H or P\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:67:3: note: previous declaration 'WebRequestMethod HTTP_HEAD'
   67 |   HTTP_HEAD    = 0b00100000,
      |   ^~~~~~~~~
In file included from e:\H or P\Documents\Arduino\libraries\ElegantOTA\src/ElegantOTA.h:57,
                 from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:34:
C:\Users\H or P\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:47:50: error: 'HTTP_POST' conflicts with a previous declaration
   47 | enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
      |                                                  ^~~~~~~~~
In file included from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:33:
e:\H or P\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:63:3: note: previous declaration 'WebRequestMethod HTTP_POST'
   63 |   HTTP_POST    = 0b00000010,
      |   ^~~~~~~~~
In file included from e:\H or P\Documents\Arduino\libraries\ElegantOTA\src/ElegantOTA.h:57,
                 from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:34:
C:\Users\H or P\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:47:61: error: 'HTTP_PUT' conflicts with a previous declaration
   47 | enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
      |                                                             ^~~~~~~~
In file included from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:33:
e:\H or P\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:65:3: note: previous declaration 'WebRequestMethod HTTP_PUT'
   65 |   HTTP_PUT     = 0b00001000,
      |   ^~~~~~~~
In file included from e:\H or P\Documents\Arduino\libraries\ElegantOTA\src/ElegantOTA.h:57,
                 from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:34:
C:\Users\H or P\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:47:71: error: 'HTTP_PATCH' conflicts with a previous declaration
   47 | enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
      |                                                                       ^~~~~~~~~~
In file included from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:33:
e:\H or P\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:66:3: note: previous declaration 'WebRequestMethod HTTP_PATCH'
   66 |   HTTP_PATCH   = 0b00010000,
      |   ^~~~~~~~~~
In file included from e:\H or P\Documents\Arduino\libraries\ElegantOTA\src/ElegantOTA.h:57,
                 from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:34:
C:\Users\H or P\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:47:83: error: 'HTTP_DELETE' conflicts with a previous declaration
   47 | enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
      |                                                                                   ^~~~~~~~~~~
In file included from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:33:
e:\H or P\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:64:3: note: previous declaration 'WebRequestMethod HTTP_DELETE'
   64 |   HTTP_DELETE  = 0b00000100,
      |   ^~~~~~~~~~~
In file included from e:\H or P\Documents\Arduino\libraries\ElegantOTA\src/ElegantOTA.h:57,
                 from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:34:
C:\Users\H or P\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer\src/ESP8266WebServer.h:47:96: error: 'HTTP_OPTIONS' conflicts with a previous declaration
   47 | enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
      |                                                                                                ^~~~~~~~~~~~
In file included from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:33:
e:\H or P\Documents\Arduino\libraries\ESP_Async_WebServer\src/ESPAsyncWebServer.h:68:3: note: previous declaration 'WebRequestMethod HTTP_OPTIONS'
   68 |   HTTP_OPTIONS = 0b01000000,
      |   ^~~~~~~~~~~~
C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino: In function 'void setup()':
C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:88:20: error: cannot convert 'AsyncWebServer*' to 'ESP8266WebServer*' {aka 'esp8266webserver::ESP8266WebServerTemplate<WiFiServer>*'}
   88 |   ElegantOTA.begin(&server);    // Start ElegantOTA
      |                    ^~~~~~~
      |                    |
      |                    AsyncWebServer*
In file included from C:\Users\H or P\AppData\Local\Temp\.arduinoIDE-unsaved202403-5520-15j8nus.1bh1\sketch_jan3a\sketch_jan3a.ino:34:
e:\H or P\Documents\Arduino\libraries\ElegantOTA\src/ElegantOTA.h:107:38: note:   initializing argument 1 of 'void ElegantOTAClass::begin(ESP8266WebServer*, const char*, const char*)'
  107 |     void begin(ELEGANTOTA_WEBSERVER *server, const char * username = "", const char * password = "");
      |                                      ^

exit status 1

Compilation error: cannot convert 'AsyncWebServer*' to 'ESP8266WebServer*' {aka 'esp8266webserver::ESP8266WebServerTemplate<WiFiServer>*'}

ayushsharma82, 2024-01-07:
This usually happens when you are trying to use both webserver ( sync and async ) in the same code.

You will have to choose between sync or async type for your whole application to resolve the conflict.

HECSmith, 2024-01-07:
Thanks for replying quickly.  

(Sigh).  It is such a 'RTFM' omission I made.  It was indeed the source of my problem and now compiles fine.  Thanks both for the help and the effort to make/maintain this library.

closed 2024-01-07 by HECSmith

gpena208777, 2024-08-31:
> Thanks for replying quickly.
> 
> (Sigh). It is such a 'RTFM' omission I made. It was indeed the source of my problem and now compiles fine. Thanks both for the help and the effort to make/maintain this library.

How did you fix this?