ayushsharma82/ElegantOTA #169: no matching function for call to 'ElegantOTAClass::begin(AsyncWebServer*)'

StefanL38, 2024-03-09:
I get a compiler error
no matching function for call to 'ElegantOTAClass::begin(AsyncWebServer*)'

what is wrong?
Is this a problem because I have used the older AsyncElegantOTA?

Arduino: 1.8.19 (Windows 10), TD: 1.58, Board: "ESP32 Dev Module, Disabled, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None, Disabled"

Using board 'esp32' from platform in folder: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14

Using core 'esp32' from platform in folder: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14

cmd /c if exist "C:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\sketchbook\\Async-OTA-Demo-001\\partitions.csv" COPY /y "C:\\Arduino-Pure-Portable\\arduino-1.8.19\\portable\\sketchbook\\Async-OTA-Demo-001\\partitions.csv" "C:\\Users\\dipl-\\AppData\\Local\\Temp\\arduino_build_776315\\partitions.csv"

esptool.py v4.5.1

Creating esp32 image...

Merged 1 ELF section

Successfully created esp32 image.

cmd /c type nul > "C:\\Users\\dipl-\\AppData\\Local\\Temp\\arduino_build_776315/file_opts"

Detecting libraries used...

Alternatives for WiFi.h: [WiFiNINA@1.8.14 WiFi@2.0.0 WiFi@1.2.7]

ResolveLibrary(WiFi.h)

  -> candidates: [WiFiNINA@1.8.14 WiFi@2.0.0 WiFi@1.2.7]

Alternatives for AsyncTCP.h: [AsyncTCP@1.1.4]

ResolveLibrary(AsyncTCP.h)

  -> candidates: [AsyncTCP@1.1.4]

Alternatives for ESPAsyncWebServer.h: [ESPAsyncWebServer@1.2.3]

ResolveLibrary(ESPAsyncWebServer.h)

  -> candidates: [ESPAsyncWebServer@1.2.3]

Alternatives for FS.h: [FS@2.0.0]

ResolveLibrary(FS.h)

  -> candidates: [FS@2.0.0]

Alternatives for ElegantOTA.h: [ElegantOTA@3.1.0]

ResolveLibrary(ElegantOTA.h)

  -> candidates: [ElegantOTA@3.1.0]

Alternatives for Update.h: [Update@2.0.0]

ResolveLibrary(Update.h)

  -> candidates: [Update@2.0.0]

Alternatives for WebServer.h: [WebServer@2.0.0]

ResolveLibrary(WebServer.h)

  -> candidates: [WebServer@2.0.0]

Generating function prototypes...

[... 258 lines cut ...]
Multiple libraries were found for "WiFi.h"

 Used: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\WiFi

 Not used: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\sketchbook\libraries\WiFiNINA

 Not used: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\sketchbook\libraries\WiFi

Using library WiFi at version 2.0.0 in folder: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\WiFi 

Using library AsyncTCP at version 1.1.4 in folder: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\sketchbook\libraries\AsyncTCP 

Using library ESPAsyncWebServer at version 1.2.3 in folder: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\sketchbook\libraries\ESPAsyncWebServer 

Using library FS at version 2.0.0 in folder: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\FS 

Using library ElegantOTA at version 3.1.0 in folder: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\sketchbook\libraries\ElegantOTA 

Using library Update at version 2.0.0 in folder: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\Update 

Using library WebServer at version 2.0.0 in folder: C:\Arduino-Pure-Portable\arduino-1.8.19\portable\packages\esp32\hardware\esp32\2.0.14\libraries\WebServer 

exit status 1

no matching function for call to 'ElegantOTAClass::begin(AsyncWebServer*)'

StefanL38, 2024-03-09:
I found the solution this modifying of the file ElegantOTA.h.

This modification is far from beeing userfriendly

You should find a way that this will be adjusted through a 
#define 
inside your demo.**ino**-file 
or 
as a variant of the constructor with a parameter inside the demo.**ino**-file 

best regards Stefan

mathieucarbou, 2024-03-10:
> You should find a way that this will be adjusted through a
> #define
> inside your demo.**ino**-file

There are 2 ways to fix this:

1) re-order the includes in order to put ESPAsyncWebServer at the end
2) If you use ESPAsyncWebServer, use async mode: https://docs.elegantota.pro/async-mode/

ednieuw, 2024-03-27:
I hope you solve the error with this define. 
" Locate the ELEGANTOTA_USE_ASYNC_WEBSERVER macro in the ElegantOTA.h file, and set it to 1:   #define ELEGANTOTA_USE_ASYNC_WEBSERVER 1"

After een upgrade to a newer version the define is overwritten and you get a compile error

thehans, 2024-06-19:
Requiring the user to manually edit a dependency (this library) for it to compile is awful.  I guess I will try the obsolete `AsyncElegantOTA` instead.

mathieucarbou, 2024-06-19:
> Requiring the user to manually edit a dependency (this library) for it to compile is awful. I guess I will try the obsolete `AsyncElegantOTA` instead.

no need to edit. 

just set a build flag: `build_flags=-DELEGANTOTA_USE_ASYNC_WEBSERVER=1`

StefanL38, 2024-07-24:
@mathieucarbou 
no need to edit.

just set a build flag: build_flags=-DELEGANTOTA_USE_ASYNC_WEBSERVER=1
Not everybody has the same knwoeldge-level as you

can you please explain in detail in which file at which line do I set this build-flag?

ayushsharma82, 2024-07-24:
@StefanL38 just the matter of reading docs: https://docs.elegantota.pro/async-mode/#for-platformio

StefanL38, 2024-07-24:
Directly quoting what is written at the link 

`
Enabling Async Mode For Arduino IDE

Go to your Arduino libraries directory
Open ElegantOTA folder and then open src folder
Locate the ELEGANTOTA_USE_ASYNC_WEBSERVER macro in the ElegantOTA.h file, and set it to 1:
#define ELEGANTOTA_USE_ASYNC_WEBSERVER 1
Save the changes to the ElegantOTA.h file.

You are now ready to use ElegantOTA in async mode for your OTA updates, utilizing the ESPAsyncWebServer library.
For PlatformIO
If you use PlatformIO then we can use a neat trick! You can set build_flags in your platformio.ini file that enables Async Mode of ElegantOTA.

Open platformio.ini file of your project
Paste this line build_flags=-DELEGANTOTA_USE_ASYNC_WEBSERVER=1 at bottom
Done!
PlatformIO will now compile ElegantOTA in Async Mode!

Benefits of Async Mode
Enabling async mode for ElegantOTA offers several advantages:

Non-blocking Operation: In async mode, ElegantOTA leverages the asynchronous web server capabilities of ESPAsyncWebServer. This means that OTA updates won't block your device's main loop. It allows your device to continue executing other tasks while handling OTA updates in the background.

Improved Responsiveness: Your device can remain responsive to user interactions or other processes during OTA updates, ensuring a smoother user experience.
`

mathieucarbou, 2024-07-24:
Example if you use PlatformIO (But in this case, it would have been better to read the doc about how to set build flags with PlatformIO).

[image]

The doc explains one way to do it. 
But setting build flags is common in C/C++ and specific to the IDE. This is a way to change the program behaviour from an external config flag. So it is up to the user to know how to set those by reading the manual of its IDE. A library cannot document all the possible ways to set flags for each IDE: this is  completely impossible.

And It does not require any specific knowledge level - this is part of C/C++ basics about how to compile that anyone should know when dealing with C / C++ libraries. Arduino IDE has a doc about that, PlatformIO also, and @ayushsharma82 also added another alternative on his website, so it is more than enough IMO.

closed 2024-08-06 by ayushsharma82