ayushsharma82/ElegantOTA #176: platformio_upload.py with authentication not working

JC-Electronics-Design, 2024-04-15:
When I configure authentication the upload with platformio_upload.py script is not working. When using the script without authentication everything works fine. 

The first thing I noticed in the script is that `checkAuthResponse = requests.get(f"{upload_url_compatibility}/update")` should be (I think) `checkAuthResponse = requests.get(f"{upload_url_compatibility}")` otherwise you end up with a URL ending with `/upload/upload` since `upload_url_compatibility` already contains `/upload`. 

When using the original script I get the following lines in the log:
```
Serverconfiguration: autentication not needed.
start-request faild 401
```
When I remove `/upload` the script tries to authenticate with `doUpdateAuth = requests.get(start_url, headers=start_headers, auth=auth)` but it fails with status code `401` again.
```
Serverconfiguration: authentication needed.
authentication faild 401
```
I did configure the correct username and password in the platformio.ini file with `custom_username` and `custom_password`. 
platformio.ini file:
```
extra_scripts = platformio_upload.py
upload_protocol = custom
custom_upload_url = http://192.168.0.104/update
custom_username = test
custom_password = test
```
ElegantOTA setup:
```
  // Set Authentication Credentials
  ElegantOTA.setAuth("test", "test");
  // Start ElegantOTA
  ElegantOTA.begin(&server);
  // ElegantOTA callbacks
  ElegantOTA.onStart(onOTAStart);
  ElegantOTA.onProgress(onOTAProgress);
  ElegantOTA.onEnd(onOTAEnd);
```
I suppose it has something to do with one of the following lines
```
auth = HTTPDigestAuth(username, password)
doUpdateAuth = requests.get(start_url, headers=start_headers, auth=auth)
```
but I'm not sure.

Anyone else that has this issue? Or is this issue known?

mleroy, 2024-04-21:
I had the same symptoms. To work around the script's bug, make sure to specify `custom_upload_url` without the `/upload` part, and without a trailing `/` as well (this would lead to a 500 error).

When I was left with a 401 error, it was because I was providing my username/password values wrapped in single quotes. 

i.e. wrong:
```
custom_upload_url=http://192.168.x.x/
custom_username='my_username'
custom_password='my_password'
```

good:

```
custom_upload_url=http://192.168.x.x
custom_username=my_username
custom_password=my_password
```

JC-Electronics-Design, 2024-04-23:
@mleroy removing the `/update` part from the custom url did the trick. Thanks!

@ayushsharma82 is this something that can be updated/added in the documentation?

closed 2024-07-06 by github-actions[bot]