ayushsharma82/ElegantOTA #177: no <wifi.h> & <update.h> files found.

Ugobyte, 2024-04-16:
I have an ESP32-S3-Dev kit and I'm trying to incoporate ElegantOTA, I'm using VSCode and PlatoformIO.
After adding these to my platform.ini `lib_deps =
  bblanchon/ArduinoJson @ 7.0.3
  esphome/AsyncTCP-esphome @ 2.1.3
  mathieucarbou/ESP Async WebServer @ 2.7.0
  https://github.com/ayushsharma82/ElegantOTA`

I get errors saying no <wifi.h> nor <update.h> found.
Can some one help me out  here?
Thanks

[image]

mathieucarbou, 2024-04-17:
Hello,
Try: https://docs.elegantota.pro/async-mode/
If it is not working, please try reproduce and isolate the error with one of the samples + your platfomio file and send the pio config file ;-)

Ugobyte, 2024-04-19:
> Hello, Try: https://docs.elegantota.pro/async-mode/ If it is not working, please try reproduce and isolate the error with one of the samples + your platfomio file and send the pio config file ;-)

Hello thanks for replying.
I tried your suggestion and I'm still getting errors.

This is my platform.ini
```
[env:esp32-s3-devkitc-1]
platform = https://github.com/platformio/platform-espressif32.git#f6ec3926f9f660ee9abada8540ffe1e205da4bbf
board = esp32-s3-devkitc-1
board_build.mcu = esp32s3
board_build.f_cpu = 240000000L
framework = arduino
monitor_speed = 115200
debug_tool = esp-prog
debug_init_break = tbreak setup
platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#44da992b774f76777bb2e931dd76cfcf12b9fe70
lib_deps =
  bblanchon/ArduinoJson @ 7.0.3
  esphome/AsyncTCP-esphome @ 2.1.3
  mathieucarbou/ESP Async WebServer @ 2.7.0
  https://github.com/ayushsharma82/ElegantOTA
build_flags=-DELEGANTOTA_USE_ASYNC_WEBSERVER=1
```

Ugobyte, 2024-04-19:
I got it to work.

I added 
```
lib_ldf_mode = deep+
```
to my platform.ini

closed 2024-04-19 by Ugobyte