ayushsharma82/ElegantOTA #182: Async Demo will not compile

Tsjippy, 2024-05-01:
The demoe supplied with this lib will not compile with error: AsyncDemo.ino:33:10: fatal error: ESPAsyncWebServer.h: No such file or directory.

And indeed that file does not exist in this repo

mathieucarbou, 2024-05-02:
This is a dependency - read the doc.

#178 might help you also once merged.

JackV2020, 2024-07-01:
I have the same issue on a Wemos D1 mini Pro which is ESP8266 based.
The Demo compiles fine but the AsyncDemo does not.
I added '#define ELEGANTOTA_USE_ASYNC_WEBSERVER 1' before the include '#include <ElegantOTA.h>' but that does not fix it. I get other compile errors. Please have a look at it because I still use AsyncElegantOTA which asks me to change to ElegantOTA.

mathieucarbou, 2024-07-01:
Hello,
Please make sure to upgrade your dependencies, especially ESPAsyncWebServer: there has been breaking changes in v3.
https://github.com/mathieucarbou/ESPAsyncWebServer/releases

JackV2020, 2024-07-01:
I used 2.10.8 until yesterday and now I have 3.0.6 of today. I changed the 'const things' and the '_P things' as the compiler messages noted. Now I thought let me try to update the AsyncElegantOTA to ElegantOTA and the basic Demo compiles but the AsyncDemo does not. When I add the "#define ELEGANTOTA_USE_ASYNC_WEBSERVER 1" line to the top of AsyncDemo it does not complain about the type anymore but I get : 
/home/jack/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino/sketches/B385A77D77EF8B89A60E1E900769B9FD/sketch/AsyncDemo.ino.cpp.o:(.text.setup+0x58): undefined reference to `_ZN15ElegantOTAClass5beginEP14AsyncWebServerPKcS3_'
/home/jack/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld: /tmp/arduino/sketches/B385A77D77EF8B89A60E1E900769B9FD/sketch/AsyncDemo.ino.cpp.o: in function `setup':
/tmp/.arduinoIDE-unsaved202461-7672-11io1pu.7x5uf/AsyncDemo/AsyncDemo.ino:74: undefined reference to `_ZN15ElegantOTAClass5beginEP14AsyncWebServerPKcS3_'
collect2: error: ld returned 1 exit status
Multiple libraries were found for "Hash.h"
  Used: /home/jack/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/Hash
  Not used: /Data/Nextcloud/Jack/Wemos/02. projects/Arduino/libraries/AsyncElegantOTA
exit status 1

Compilation error: exit status 1

mathieucarbou, 2024-07-01:
It looks like an environment issue. The CI pass and pio build pass with the mentinned combination.

JackV2020, 2024-07-01:
I removed my Arduino IDE setup and recreated it and still have the issue. Please read the below.

I use Arduino IDE 2.3.2 AppImage version.
I exited the IDE and removed the next folders :

/home/jack/.arduino15
/home/jack/.arduinoIDE
/home/jack/Arduino
/home/jack/.config/Arduino IDE/
/home/jack/.config/arduino-ide/

renamed my projects folder to keep my stuff and created a new one with the name it had before.
created a subfolder libraries in it
started arduino-ide_2.3.2_Linux_64bit.AppImage and had to accept the 'Terms of Service'.
No examples installed, no libraries and all settings are gone....
Changed settings : Sketbook location to projects folder and added the board manager url for Wemos :
  https://arduino.esp8266.com/stable/package_esp8266com_index.json
Installed the board manager for ESP8266
selected my serial port /dev/ttyACM0
selected Wemos D1 mini Pro
tested the blink example
installed 'ESP Async WebServer' 3.0.6
installed 'ElegantOTA' 3.1.2
compiled ElegantOTA AsyncDemo and got an error on a missing library.
installed 'ESPAsyncTCP' 1.2.4
compiled ElegantOTA AsyncDemo and got a lot of errors again explaining there is something wrong with "ElegantOTA.begin(&server);" in setup() :

( there is also more I did  after this compiler output so please scroll down )

In file included from /Data/Nextcloud/Jack/Wemos/02. projects/Arduino/libraries/ElegantOTA/src/ElegantOTA.h:57,
                 from /tmp/.arduinoIDE-unsaved202461-16722-1vfm572.jyk5j/AsyncDemo/AsyncDemo.ino:34:
/home/jack/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WebServer/src/ESP8266WebServer.h:47:19: error: 'HTTP_ANY' conflicts with a previous declaration
   47 | enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
      |                   ^~~~~~~~
In file included from /tmp/.arduinoIDE-unsaved202461-16722-1vfm572.jyk5j/AsyncDemo/AsyncDemo.ino:33:
/Data/Nextcloud/Jack/Wemos/02. projects/Arduino/libraries/ESP_Async_WebServer/src/ESPAsyncWebServer.h:81:5: note: previous declaration 'WebRequestMethod HTTP_ANY'
   81 |     HTTP_ANY = 0b01111111,
      |     ^~~~~~~~
In file included from /Data/Nextcloud/Jack/Wemos/02. projects/Arduino/libraries/ElegantOTA/src/ElegantOTA.h:57,
                 from /tmp/.arduinoIDE-unsaved202461-16722-1vfm572.jyk5j/AsyncDemo/AsyncDemo.ino:34:
/home/jack/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WebServer/src/ESP8266WebServer.h:47:29: error: 'HTTP_GET' conflicts with a previous declaration
   47 | enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
      |                             ^~~~~~~~
In file included from /tmp/.arduinoIDE-unsaved202461-16722-1vfm572.jyk5j/AsyncDemo/AsyncDemo.ino:33:
/Data/Nextcloud/Jack/Wemos/02. projects/Arduino/libraries/ESP_Async_WebServer/src/ESPAsyncWebServer.h:74:5: note: previous declaration 'WebRequestMethod HTTP_GET'
   74 |     HTTP_GET = 0b00000001,
      |     ^~~~~~~~
In file included from /Data/Nextcloud/Jack/Wemos/02. projects/Arduino/libraries/ElegantOTA/src/ElegantOTA.h:57,
                 from /tmp/.arduinoIDE-unsaved202461-16722-1vfm572.jyk5j/AsyncDemo/AsyncDemo.ino:34:
/home/jack/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WebServer/src/ESP8266WebServer.h:47:39: error: 'HTTP_HEAD' conflicts with a previous declaration
   47 | enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
      |                                       ^~~~~~~~~
In file included from /tmp/.arduinoIDE-unsaved202461-16722-1vfm572.jyk5j/AsyncDemo/AsyncDemo.ino:33:
/Data/Nextcloud/Jack/Wemos/02. projects/Arduino/libraries/ESP_Async_WebServer/src/ESPAsyncWebServer.h:79:5: note: previous declaration 'WebRequestMethod HTTP_HEAD'
   79 |     HTTP_HEAD = 0b00100000,
      |     ^~~~~~~~~
In file included from /Data/Nextcloud/Jack/Wemos/02. projects/Arduino/libraries/ElegantOTA/src/ElegantOTA.h:57,
                 from /tmp/.arduinoIDE-unsaved202461-16722-1vfm572.jyk5j/AsyncDemo/AsyncDemo.ino:34:
/home/jack/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WebServer/src/ESP8266WebServer.h:47:50: error: 'HTTP_POST' conflicts with a previous declaration
   47 | enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
      |                                                  ^~~~~~~~~
In file included from /tmp/.arduinoIDE-unsaved202461-16722-1vfm572.jyk5j/AsyncDemo/AsyncDemo.ino:33:
/Data/Nextcloud/Jack/Wemos/02. projects/Arduino/libraries/ESP_Async_WebServer/src/ESPAsyncWebServer.h:75:5: note: previous declaration 'WebRequestMethod HTTP_POST'
   75 |     HTTP_POST = 0b00000010,
      |     ^~~~~~~~~
In file included from /Data/Nextcloud/Jack/Wemos/02. projects/Arduino/libraries/ElegantOTA/src/ElegantOTA.h:57,
                 from /tmp/.arduinoIDE-unsaved202461-16722-1vfm572.jyk5j/AsyncDemo/AsyncDemo.ino:34:
/home/jack/.arduino15/packages/esp8266/hardware/esp8266/3.1.2/libraries/ESP8266WebServer/src/ESP8266WebServer.h:47:61: error: 'HTTP_PUT' conflicts with a previous declaration
   47 | enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
      |                                                             ^~~~~~~~
[... 40 lines cut ...]
      |                                      ^

exit status 1

[... cut ...]
/tmp/.arduinoIDE-unsaved202461-16722-1vfm572.jyk5j/AsyncDemo/AsyncDemo.ino:74: undefined reference to `_ZN15ElegantOTAClass5beginEP14AsyncWebServerPKcS3_'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

There is a reference to setup in AsyncDemo again.

When I leave "#define ELEGANTOTA_USE_ASYNC_WEBSERVER 1" in and put comment markers before ElegantOTA.begin(&server);    // Start ElegantOTA
it compiles without issues.

I hope this helps you helping me

JackV2020, 2024-07-02:
I have it working but have a new question at the end of this.

I should have read the header of the example and go to the internet to find I have to change a line in the library itself.
Internet says:

For Arduino IDE

Go to your Arduino libraries directory

Open ElegantOTA folder and then open src folder

Locate the ELEGANTOTA_USE_ASYNC_WEBSERVER macro in the ElegantOTA.h file, and set it to 1:

Changed it in the file and AsyncDemo compiles:

#ifndef ELEGANTOTA_USE_ASYNC_WEBSERVER
  #define ELEGANTOTA_USE_ASYNC_WEBSERVER 1
#endif

So now my question why can I not leave it at 0 and put "#define ELEGANTOTA_USE_ASYNC_WEBSERVER 1" in my app like:

#define ELEGANTOTA_USE_ASYNC_WEBSERVER 1
#include <ElegantOTA.h>

I expected that to work since I wonder now what happens when I receive an update of the library.
Do I need to change the code in the library after every update of the library?

mathieucarbou, 2024-07-02:
You wouldn't have all these questions if you were using a real dev environment like Vscode with the PIO extension ;-)

ayushsharma82, 2024-07-02:
@JackV2020 It's just how Arduino IDE works and fails to implement build flags since ages. You should be fine upon switching to PIO and using the build flags to set this macro.

JackV2020, 2024-07-02:
@mathieucarbou @ayushsharma82  Indeed, I have Arduino IDE and use the AppImage version of it and have no experience with visual studio or platform io or the combination of them but I did put that on my to do list today. 
Last days I have been busy with this ESP8266 project in which I put some handy functionality together. The idea is to have a 'central server' in the IoT network. It has things like Web access to the LittleFS so I can create files and directory trees, upload and download files, create files and also **edit files in a web editor**. There is a web socket application in it with examples for a sort of MQTT like server and a web socket server to control and react on interrupt of a number of GPIO's One other thing in it is a NAPT and reverse NAT router to have a little dedicated WiFi for some IoT things. (Dhcp and Reverse NAT settings can be editted with the LittleFS Web editor ) You may have a look at the readme on https://github.com/JackV2020/ESP8266-Server which also contains screenshots of what it looks like. At this moment that version still has the old OTA solution but I will put the new one in. 
Many thanks again. Jack

JackV2020, 2024-08-02:
I did not put the new one in. I created a customizable one: https://github.com/JackV2020/ESP8266-Firmware-Over-The-Air

HenkHoldijk, 2024-08-04:
Just tried to compile the AsyncDemo.ino

Arduino IDE 2.3.2
arduino-esp32 3.0.4
Board ESP32-C3 Super Mini

The example is definitely broken and does not use Async by default (which I would have expected.

Instead of using the define, I think it would be wiser to add an additional (abstraction) layer of include files. One for the normal "ElegantOTA" and one for the "ElegantAsyncOTA". This also avoids the problems caused by the Arduino IDE shuffling the order of things while creating the actual code that is compiled. In these 2 include files all required defines / typedefs or even different class definitions can be made.

ayushsharma82, 2024-08-05:
@HenkHoldijk The async demo example tells you to enable the Async mode beforehand: https://github.com/ayushsharma82/ElegantOTA/blob/ecdf53f0ace6c8e2acd3902cdc14ab87b538d0fe/examples/AsyncDemo/AsyncDemo.ino#L6

The macro route was decided after a lot of factors, it's the only way to make a project easily maintainable which shares common code.

Yes, slight inconvenience of setting the flag when it comes to Arduino IDE but that's really a disability of Arduino IDE itself. Setting build flags is the most basic and fundamental thing found in C/C++ IDEs but they opted for not adding it. Hopefully, it will be added soon, until then this is the way forward.

closed 2024-08-05 by ayushsharma82

HenkHoldijk, 2024-08-05:
Just wanted to let you know that even with the "#define ELEGANTOTA_USE_ASYNC_WEBSERVER 1" in top of the sketch, it does not compile. That is why I said that it was "definitely broken". IDE 2.3.2. & ESP32 core 3.0.4 & Board ESP32C3 Super Mini.

ayushsharma82, 2024-08-05:
Ref: https://docs.elegantota.pro/async-mode/#enabling-async-mode

PearlDragon, 2025-03-14:
The library files are readonly in Arduino IDE